Ultimaker/Cura #9917: Unused Extruder in PrimeTower

EllVanCraft, 2021-05-27:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Custom FFF

### Reproduction steps

1. I am running Cura 4.9.1 on a Windows 10
2. Select my Custom FFF printer with 8 extruders
2. Added 3 cubes into cura
3. CubeA receives Extruder 1, CubeB receives Extruder 2, CubeC receives Extruder 3
4.  Enabled Dual Extrusion with Prime Tower
5. The generated Prime Tower shows that Extruder 5 will purge on every layer but there is no object with Extruder 5

### Actual results

The generated Prime Tower shows that Extruder 5 will purge on every layer but there is no object with Extruder 5

### Expected results

I expect that only the used extruders are purging in the Prime Tower

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I tried the same on a Linux system and Cura 4.8. The issue was not reproducible.
I had the issue on a Windows 10 with Cura 4.9.0 and 4.9.1

fvrmr, 2021-05-28:
Hi @EllVanCraft thank you for your report.
Could you share your project file? (file - save project) I can't reproduce this issue with your reproduction steps.

EllVanCraft, 2021-05-29:
Hello Fenne,

I attached you the cura file and the gcode file. You will see that there is a tool T4 in the gcode which was never assigned in cura.

Have a nice weekend,
Daniel

From: Fenne
Sent: May 28, 2021 2:50 AM
To: Ultimaker/Cura
Cc: EllVanCraft; Mention
Subject: Re: [Ultimaker/Cura] Unused Extruder in PrimeTower (#9917)

Hi @EllVanCraft thank you for your report.
Could you share your project file? (file - save project) I can't reproduce this issue with your reproduction steps.

fieldOfView, 2021-05-31:
Github strips attachments sent via email. Please either post a link to something that we can download, or use the webinterface at https://github.com/Ultimaker/Cura/issues/9917 to attach the files.

EllVanCraft, 2021-06-02:
Sorry for the delay. I attached a zip file with the cura file and the gcode file. 
[cubeTest.zip](https://github.com/Ultimaker/Cura/files/6581928/cubeTest.zip)

fvrmr, 2021-06-03:
Hi @EllVanCraft thank you for your file.
I can reproduce your issue and will discuss this with the team. Keep you posted!

fvrmr, 2021-06-08:
I have created a ticket on our backlog for this. 
Devs see CURA-8298

fvrmr, 2022-02-25:
This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/8213 which was recently closed because the ticket is removed on our backlog.
I will close this issue and mark it as a duplicate.

closed 2022-02-25 by fvrmr