Ultimaker/Cura #9918: Does not slice

driedeker, 2021-05-27:
### Application Version

4.9.1

### Platform

windows 7 64

### Printer

CR10s pro

### Reproduction steps

load up a any stl and ask it to slice

### Actual results

sits there for hours and never finish slicing

### Expected results

should slice and then I can print it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Can not give you anymore info as I uninstalled it and went back to 4.8

cociv27, 2021-05-27:
Exactly the same issue on my MacOS 11.2.3 with Cura on 4.9.1.
The blue slicing bar stops and disappears.

fvrmr, 2021-05-28:
Hi @driedeker thank you for your report.
Could you share your log file and stl? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
And does this happen only with one stl or with any model you try to slice?

driedeker, 2021-05-28:
Happens with all stls
reinstalled to produce a log
[cura.log](https://github.com/Ultimaker/Cura/files/6563525/cura.log)

[Hemera_BLTouch-2-3mm.zip](https://github.com/Ultimaker/Cura/files/6563530/Hemera_BLTouch-2-3mm.zip)

fvrmr, 2021-05-31:
Hi @driedeker thank you, it looks like you don't have access to internet.
Do you have a firewall enabled?

driedeker, 2021-05-31:
How strange yes I run the firewall and it had blocked this version of cura but none of the pryer  versions

put in an exeption and now slices so now to test it out thanks.

Terry

closed 2021-06-01 by nallath

fvrmr, 2021-06-01:
Hi @driedeker thank you for letting me know. Good to know that it is working!