Ultimaker/Cura #9919: "create profile from current settings / overrides" causes all settings to be lost and no profile is saved

NateCroix, 2021-05-28:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Custom Ender3 SKR1.4 Marlin

### Reproduction steps

- Create settings overrides
- Slice
- Click "create profile from current settings / overrides"
- Name File
- Click "Save"

### Actual results

- all override values disappear but custom visibility of settings override remain in main settings menu scroll.
All custom values in these overrides are reset, no presets are saved. Searching for the preset name in file folders yields no results
- Wont let save any custom profiles

### Expected results

Profile should be saved and able to be recalled

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Video showing the actions and result.](https://streamable.com/8ewsxi)

Logfile is not timestamped with anything near the time I had this issue

NateCroix, 2021-05-28:
apparently if any material type preset has been chosen, the normal profile saving mechanism does not work. Figured out how to see profiles that were saved by creating a custom material and selecting that.

closed 2021-05-28 by NateCroix

Ghostkeeper, 2021-06-07:
Hmm, that would still sound like a problem to me. The profile shouldn't disappear immediately after creating it.

It should disappear if the printer has material-specific profiles (like yours) and you change the material to something with a different Material Type. But that's not what you did in the video.

However I also can't reproduce your steps by repeating the steps. For me the profiles stay visible. I tried it with eSUN PLA+ (which is of unsupported material type "PLA+" for the Ender 3). And I tried it with a custom material. Both seemed to work fine, letting me create profiles and keeping those profiles visible too.