Ultimaker/Cura #9920: Program completely crashes when trying to open a file

imLokiGod, 2021-05-28:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Anycubic Chiron

### Reproduction steps

After booting up, tried to open a file, program crashes. 9 out of 10 times, the File box that opens is completely white, and then crashes. 
 Tried restarting program, reinstalling etc. nothing is working. Cura worked fine until yesterday when this started. 
5 month old gaming computer, so not a hardware issue. have done complete virus scans etc to rule anything out. 

### Actual results

File box opens, then Cura crashes

### Expected results

It should stay open and work lol

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No other info to provide. Straight forward crashing

LordStarhawk, 2021-05-28:
I am experiencing the same issue.
[cura.log](https://github.com/Ultimaker/Cura/files/6558855/cura.log)

fvrmr, 2021-05-28:
Hi @imLokiGod thank you for your report.
This is probably a duplicate of https://github.com/Ultimaker/Cura/issues/9668 
There you can find a solution that is related to Chitubox or you need to update plugins. 

I will close this issue since it is a duplicate. But please reach out if you have more questions.

closed 2021-05-28 by fvrmr