Ultimaker/Cura #9921: cant connect to printer, cant log in to ultimaker

al6600, 2021-05-28:
### Application Version

4.9.1

### Platform

windows 10 intel(r) core(tm) i5-8400 cpu 2.8ghz 2.81 ghz ram 16gb

### Printer

ultimaker 3

### Reproduction steps

tried to connect to printer via the add printer menu, printer does not show up.
tried to add with ip address, nothing happened after pressing ok.
tried to sign into ultimaker account to see if it would add it, just get the round working symbol and nothing happens, have to force close cura.

### Actual results

have to go back to older version 4.8.0 everything works here

### Expected results

should see printer and be able to send jobs directly from pc, should be able to monitor on ultimaker connect

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6558975/cura.log)

fvrmr, 2021-05-28:
Hi @al6600 thank you for your bug report.
I will discuss this with the team, keep you posted!

fvrmr, 2021-05-31:
Hi @al6600 do you have a firewall enabled?

al6600, 2021-05-31:
Hi yes I have bullguard installed

fvrmr, 2021-05-31:
Could you try to disable your firewall and see if it is disabled you can connect your printer?

al6600, 2021-06-01:
Hi I disabled the firewall and its working fine now. Strange because it
doesnt affect older cura.

al6600, 2021-06-01:
Just to add, I switched the firewall on again with cura still open and then
the firewall asked for permission to allow cura to access the Internet. It
didn't do this before and now works with the firewall on

fvrmr, 2021-06-01:
Hi @al6600 thanks for letting me know. And good to know that Cura is working now with the firewall.
I will close this issue since it is solved!

closed 2021-06-01 by fvrmr

al6600, 2021-06-01:
Thanks for your help