Ultimaker/Cura #9922: Crash on startup?

iceguru, 2021-05-28:
### Application Version

4.9.1

### Platform

windows 10

### Printer

none yet

### Reproduction steps

I started the program multiple times and each time it crashed right after the white screen began.

### Actual results

Unsure what's being asked of me with the "above steps"

### Expected results

It should load?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

`trimesh\util.py:2064: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.9 it will stop working
networkx\readwrite\graphml.py:346: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
networkx\readwrite\gexf.py:220: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
trimesh\voxel\runlength.py:205: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
2021-05-28 03:22:07,230 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ConsoleLogger
2021-05-28 03:22:07,280 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraDrive
2021-05-28 03:22:07,354 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [96]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.9.1\CuraEngine.exe
2021-05-28 03:22:07,365 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraEngineBackend
2021-05-28 03:22:07,391 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraProfileReader
2021-05-28 03:22:07,399 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2021-05-28 03:22:07,413 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.9.1\CuraEngine.exe
2021-05-28 03:22:07,421 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.9.1\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2021-05-28 03:22:07,448 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraProfileWriter
2021-05-28 03:22:07,565 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin DigitalLibrary
2021-05-28 03:22:07,590 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin OBJReader
2021-05-28 03:22:07,612 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin OBJWriter
2021-05-28 03:22:07,757 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin STLReader
2021-05-28 03:22:07,784 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin STLWriter
2021-05-28 03:22:07,808 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-05-28 03:22:07,844 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-05-28 03:22:07,873 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-05-28 03:22:07,895 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-05-28 03:22:07,918 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-05-28 03:22:07,945 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-05-28 03:22:08,056 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-05-28 03:22:08,065 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.9.1
2021-05-28 03:22:08,072 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2021-05-28 03:22:08,078 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-05-28 03:22:08,087 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-05-28 03:22:08,095 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-05-28 03:22:08,102 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-05-28 03:22:08,120 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-05-28 03:22:08,129 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-05-28 03:22:08,137 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-05-28 03:22:08,145 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-05-28 03:22:08,154 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
[... 427 lines cut ...]

Thread 0x00003bf0 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
 xe\build\inst\lib\python3.8\site-packages\UM\View\GL\OpenGL.py", line 197 in createShaderProgram
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\SolidView\SolidView.py", line 156 in _checkSetup
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\SolidView\Solidackages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
 mp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\bin\cura_app.py",ython3.8\site-packages\UM\View\GL\ShaderProgram.py", line 95 in load
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\View\GL\OpenGL.py", line 197 in createShaderProgram
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\SolidView\SolidView.py", line 156 in _checkSetup
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\SolidView\SolidView.py", line 212 in beginRendering
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\Bindings\MainWindow.py", line 238 in _render
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run
`

fvrmr, 2021-05-28:
Hi @iceguru thank you for your bug report. 
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9847 there you can find a work around and there is a fix made for this.
I will close this because this is a duplicate.

iceguru, 2021-05-30:
The fix in that thread to "delete appdata" didn't work for me.

fvrmr, 2021-05-31:
Did you try this?

> removing completely `ultimaker_auth_data object` from the config file solved the issue

And the fix for this will be in the next release of Cura.

iceguru, 2021-05-31:
> Did you try this?
> 
> > removing completely `ultimaker_auth_data object` from the config file solved the issue
> 
> And the fix for this will be in the next release of Cura.

`[general]
last_run_version = 4.9.1
version = 7

[view]

[cura]

[cura_drive]

[backend]

[info]

[gcodereader]

[local_file]

[layerview]

`

That's not in my config

fvrmr, 2021-06-01:
Then you need to wait till the next release to have the fix for this. Does older versions of Cura work for you?

iceguru, 2021-06-02:
The older .7 version did work for me.

fieldOfView, 2021-06-12:
This does not look like a duplicate to #9847 to me. The crashlog is different. The mention of a white screen always reminds me of https://github.com/Ultimaker/Cura/issues/8892, but @iceguru says that removing the appdata does not fix the issue so that might not be it either. @iceguru, do you have multiple displays?

I see that @iceguru is using a rather old GPU (2010-ish). Could be a driver issue.

closed 2021-06-28 by no-response[bot]

amiazma, 2022-05-07:
Its same for me. Version: 4.13.1 and lower and beta 5.0. I tried: Win+P single monitor. Delet %Appdata. Install other drive. Run with 7zip without install. Installed previous versions. The V2.x worked. :D Its gone after starting from Task Managger. I deleted everything from registry witch contained Cura. Restarted a lot. Upgraded Win10 and drivers. Installed python 3.10, uninstalled, installed 2.x. Tried this config ting too, but there isnt this line for me too... Pls help me! :D

There are my log files (stderr and cura.log) (Deleted the old ones, so its a fresh crash. :) )

> 2022-05-07 16:14:23,729 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [83]: Adding QT5 plugin path: D:\Program Files\Ultimaker Cura 4.13.1\PyQt5\plugins
> 2022-05-07 16:14:23,730 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [93]: Adding QT5 plugin path: D:\Program Files\Ultimaker Cura 4.13.1\PyQt5\plugins
> 2022-05-07 16:14:23,730 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
> 2022-05-07 16:14:23,730 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 4.13.1
> 2022-05-07 16:14:23,730 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 7.9.0
> 2022-05-07 16:14:23,730 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
> 2022-05-07 16:14:23,730 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\Ultimaker Cura 4.13.1\share\uranium\plugins must be a folder.
> 2022-05-07 16:14:23,730 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\Ultimaker Cura 4.13.1\share\cura\plugins must be a folder.
> 2022-05-07 16:14:23,731 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\lib\uranium must be a folder.
> 2022-05-07 16:14:23,731 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\lib64\uranium must be a folder.
> 2022-05-07 16:14:23,731 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\lib32\uranium must be a folder.
> 2022-05-07 16:14:23,731 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\Resources\uranium\plugins must be a folder.
> 2022-05-07 16:14:23,731 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\Resources\cura\plugins must be a folder.
> 2022-05-07 16:14:23,731 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
> 2022-05-07 16:14:23,731 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is C:\Users\Windows\AppData\Roaming\cura\4.13
> 2022-05-07 16:14:23,731 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is C:\Users\Windows\AppData\Roaming\cura\4.13
> 2022-05-07 16:14:23,731 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is C:\Users\Windows\AppData\Local\cura\4.13\cache
> 2022-05-07 16:14:23,731 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: D:\Program Files\Ultimaker Cura 4.13.1\resources\bundled_packages\cura.json
> 2022-05-07 16:14:23,731 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: D:\Program Files\Ultimaker Cura 4.13.1\resources\bundled_packages\uranium.json
> 2022-05-07 16:14:23,732 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [123]: Trying OpenGL context 4.1...
> 2022-05-07 16:14:23,732 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [131]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
> 2022-05-07 16:14:23,732 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [177]: Detected most suitable OpenGL context version: 4.1 Core profile
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.Qt.QtApplication.initialize [184]: Initializing job queue ...
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.Qt.QtApplication.initialize [188]: Initializing version upgrade manager ...
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from D:\Program Files\Ultimaker Cura 4.13.1\resources\bundled_packages\cura.json
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from D:\Program Files\Ultimaker Cura 4.13.1\resources\bundled_packages\uranium.json
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.PackageManager._loadManagementData [197]: Loaded user packages management file from C:\Users\Windows\AppData\Roaming\cura\4.13\packages.json
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.PackageManager._saveManagementData [270]: Package management file C:\Users\Windows\AppData\Roaming\cura\4.13\packages.json was saved
> 2022-05-07 16:14:23,732 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [119]: Loading plugin configuration file 'C:\Users\Windows\AppData\Roaming\cura\4.13\plugins.json'
> 2022-05-07 16:14:23,732 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Program Files\lib\cura must be a folder.
> 2022-05-07 16:14:23,736 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin FileLogger
> 2022-05-07 16:14:23,750 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin FirmwareUpdateChecker
> 2022-05-07 16:14:23,761 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin FirmwareUpdater
[... cut ...]
>   File "threading.py", line 932 in _bootstrap_inner
>   File "threading.py", line 890 in _bootstrap
> 
> Current thread 0x00002c28 (most recent call first):
>   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 396 in exec_
>   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 877 in run
>   File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 236 in <module>
>   File "Console.py", line 36 in run
>   File "__startup__.py", line 74 in run