Ultimaker/Cura #9924: Possibility to add M92 Extrusion steps to the Material Settings and not at the Printer Settings Start-G-Code

awiltschek, 2021-05-29:
### Is your feature request related to a problem?

I need to manage my material Start G-Code settings in the printer and material selected is not impacting the slicing and settings in the prepare anyhow.

### Describe the solution you'd like

At the moment when changing Filament i need to have in the Start-G-code the measured M92 Exxxx Steps for each material.
It would be great if you implement a Start G-Code Section also to the Material which will be included automatically at the Beginning and then the Start G-Code of the printer settings.
So it would be possible to habe material related Settings and Codes at the material and Machine related Settings at the machine.
When i then select a material before slicing. This will impact my overall Ge-Code generation better.
At the moment i allways need to change the printer settings for material selection and the material selected in the Prepare Stage has NO Impact on my slicing or even on my slicing settings.
Hope i could explain what i mean.

Example of Printer Start G-Code:
; M92 X80.12 Y80.12 Z399.78 E433.00 ; PLA 3DPrima
; M92 X80.12 Y80.12 Z399.78 E448.59 ; PETG 3DPrima-Yellow
; M92 X80.12 Y80.12 Z399.78 E521.96 ; PETG M4P-Orange
; M92 X80.12 Y80.12 Z399.78 E448.54 ; PLA Renkforce White
M92 X80.12 Y80.12 Z399.78 E529.26 ; PETG M4P-Red

### Describe alternatives you've considered

keep it as it is :-) as i have a solution for that with commenting out the not used material settings E-Steps mainly

### Affected users and/or printers

All

Asterchades, 2021-05-29:
You don't actually have to specify value to Marlin (and I would suspect RRF) if you're not changing it. Those M92s only need the new E value given the X, Y, and Z values are identical, which presumably you already have set in the printer's EEPROM. It's not hurting anything to have them, but it's one less potential source of interference to track down later if you should happen to have to change one of them for any reason.

Honest question, though, what practical benefit would this offer beyond just adjusting the already-available Flow value? Obviously the resulting E values doing that would produce would be different numerically, but the resulting print - provided you give it the correct modifier - would be almost exactly identical to if you'd changed the E-steps or indeed tweaked the Marlin global flow (M221). And I say "almost exactly" as Cura does "only" calculate down to 3 decimal places (which is all Marlin and similar accept), so there's the potential of up to 0.5um of extrusion difference here or there (which is significantly less than the error already inherent to micro-stepping).

awiltschek, 2021-05-29:
no my eprom is writeprotect as my printer is default and i want to print and not to mod and play with printer engineering.

second e-steps and flow are not the same and yes you can experiment your ideal settings with flow or you measure them (e-steps) and minimal adjust them with the flow setting.
i have done both and best practices is what i wrote.

and last why do anybody need the material when their settings are totaly worthles. so cura then can remove material as its worthless.

no it would be good when material settings get used also and their configuration and settings are included in the overall slicing configuration.

so far nobody needs this material stuff and its settings, because they dont influence my slicing and g-codes.
so worthless feature.

with my proposal you can configure material and additional g-code settings and then easily switch between materials witouth having the need to reconfigure printer start g-code.

GregValiant, 2021-05-30:
@awiltschek - I don't get it.  How can your E-steps be so wildly different (almost 19%) between your PLA and your PETG?
Are the diameters totally different?  Was a single or double wall "calibration cube" involved?

When anyone tells their printer to extrude 100mm of filament, and it extrudes exactly 100mm of filament, it is calibrated for any and all filaments of the same diameter.  I don't know those filament brands you refer to, but if I ordered 1.75mm diameter filament and received a roll that was1.59mm diameter, I'd send it back with a nasty note.

awiltschek, 2021-05-30:
> @awiltschek - I don't get it.  How can your E-steps be so wildly different (almost 19%) between your PLA and your PETG?
>
> Are the diameters totally different?  Was a single or double wall [...]

Because its from 3 different companies ?
i measured them and this are the results. for petg they are close to each other as i only have one roll from 3dprima and all others are from M4P. PLA are 2 different types of 3dprima.
and no i didnt measure the diameter. 
i adjust the e-steps on each new roll and mark them in the startup g-code. done.
then i minimal adjust the flow if necessary.

yet-another-average-joe, 2021-05-31:
> 
> When anyone tells their printer to extrude 100mm of filament, and it extrudes exactly 100mm of filament, it is calibrated for any and all filaments of the same diameter.

No. It depends of the apparent diameter of the filament, and this apparent diameter depends how hard the filament is (how the extruder theeth will bite into it), how the filament yields under the wheels pressure.

Calibration is valid for one filament, and one extruder pressure adjustment only. Calibration should be done for each filament, and/or if the wheels adjustment is modified. On my side I play with flow. Furthermore, PETG prints better with some underextrusion (see Prusa profile : flow = 95%). I use 95 to 98 depending on the filament brand, 100% being for the PLA I'm using from time to time (I print PETG 99% of the time).

@awiltschek : A couple years ago, I've been playing with the filament diameter, and for some reason (lazyness ?) I stopped doing this ; it is a good workaround, because you can keep 100% as the standard value in the Material flow.

In the filament definition, reduce or increase the filament diameter. It will have the same effect as setting steps in G-Code. (just do the maths).

GregValiant, 2021-06-01:
@yet-another-average-joe , I understand.  A quick experiment seemed in order.
Using a razor blade I carefully marked a length of PLA filament at 0 and 200mm.  I extruded the filament in a single move.  Carefully measuring the piece after it passed through my extruder (aftermarket aluminum extruder on an Ender 3 Pro) the marks were 200.00mm apart.  I ran the experiment twice with PLA and once with TPU with the same results.  Different extruders will have teeth of a different profile on the drive gear, and different spring pressure.  Extruders with dual drive gears would be different as well.  
The diameter of the PLA filament prior to the test varied from 1.72mm to 1.73mm.  After the experiment it was 1.73 to 1.74mm with the increase in diameter being across the deformations of the gear teeth.  So any deformation appears to effect the diameter rather than the length and that would have no effect on the "volume" of extruded material (whereas if the length changed it would have an effect).  That was expected as (for my single drive extruder) the plastic would flow between the teeth of the drive gear much easier than trying to lengthen filament that for much of it's cross-section is unaffected by the drive wheel.
The difference in diameter (1.725 vs 1.75) would make an error of -2.84% in the volume of the extruded filament.  I think that is a significant number and since it is the diameter of a material that Cura has to work with, it would result in an under-extrusion.
The thing that often gets lost is that what we are really talking about is the ratio of {volume-of-filament-in} to {volume-of-filament-out}.  When that ratio is 1:1 then (what people refer to as) Flow is 100%.
Accurately measuring the diameter of the filament and then informing Cura of the diameter is how I do it.  That being said, if the print looks crappy I just adjust the flow on the fly.  I think the "Mark I Eyeball" is still the best calibration tool.

yet-another-average-joe, 2021-06-01:
The extruder + filament system can be compared to a belt + pulley system (with a belt not of the tooth type), it is not a rack pinion system because there is slippage and it cannot be avoided (for example because of the opposite force generated et the nozzle + model being printed, friction, etc.). Also, when calculating a belt pulley transmission, the pulley is not the only parameter : the important variable is the effective radius : usually the pulley radius + half the belt thickness (pitch line, neutral line or whatever).
You are talking about filament diameters before and after : this takes into account the plastic deformation only... There's also the elastic deformation that occurs before permanent yielding (plastic deformation) : it cannot be measured using a micrometer.
Filament out is always equal to filament in.
The slicer does not care about the filament diameter. It is used in an ideal world, for volume calculations only. Playing with filament diameter or flow %  or steps/mm is exactly the same.

Accurately measuring the filament is of no interest by itself. Different filaments with the same measured diameters will need different flow adjustments. We see that all the time. The only important thing is that a say 0.4mm width * 0.2mm height line is as near as possible as 0.4*0.2 ; it is extremely difficult to measure (a profile projector could help, and I'm sure it has been done many times by 3D printer manufacturers) ; in the end, as you wrote : the print must look fine and that's it ! Playing with flow or with diameter or with steps has the same same effect on volume really extruded. Playing with diameter is convenient because the diameter is stored with material by all slicers.

And the real flow (on the build plate) also depends on the extrusion rate, nozzle temperature, fluids characteristics, etc... 

I adjust the steps very rarely. But I adjust flow real time in Marlin from time to time in case of new filament, change in speed, crappy print, etc., and then edit the Cura profile associated with this filament ; many do that. But it would be better to play with diameter as it is mostly material dependant and there's no other parameter to play with in the material profile. Reason why the OP feature request makes sense. Copying the empirically determined  flow % in Marlin into a material profile would be easier and faster than doing Marlin_flow * filament_diameter, or Marlin_flow * steps. Probably a matter of lazyness, and probably the reason why I stopped doing it ! I just write the % in the comment section of the material, and adjust flow everytime in the Material section.

steps/mm, filament diameter,  % flow are all related to each others and can be tweaked for the same result.

In Cura, there could be a radio button with "1.75mm / 3.0mm" and a adit field with "flow %" or "steps / mm (of course, this is caricatural  ; but it would do it the same)

(sorry for the flood and my bad english, missing vocabulary doesn't help for short text)

fvrmr, 2021-06-03:
Hi @awiltschek thank you for your feature request. 
Like being set before I also think adding this wouldn't add value since flow settings could do this as well. 
Do you have the feeling that that doesn't work for you?

awiltschek, 2021-06-03:
> Hi @awiltschek thank you for your feature request. 
> 
> Like being set before I also think adding this wouldn't add value since flow settings could do this as well. 
> 
> Do you have the feeling that that doesn't work for you? 

otherwise i would not have proposed it.

Ghostkeeper, 2021-06-07:
As I understand it, you calibrated the flow rate out of your nozzle by changing the Steps Per MM firmware parameter for the E axis. This firmware parameter determines how many motor steps result in 1mm of movement of the edge of the feeder wheel (determined by the precision of the motor and the gear ratios in the feeder).

Unless you change out the motor or change the gear ratio in the feeder when you change materials, the Steps Per MM shouldn't change.

Cura does have two proper settings to do what you want though:

* Material Diameter: Indicating how much filament movement results in how much volume of plastic extruded.
* Flow: General compensation method, but most often used to compensate for *slip*; how much the feeder wheel needs to turn for a certain amount of filament movement.

Flow, diameter and Steps Per MM all have the same end result: A scale on the number of E steps that the printer will instruct. The diameter can be set in the Material Manager window. The Flow setting not by default, but the Material Settings plug-in from the Marketplace will add that for you.

I currently think that your real problem is slip, and that you should use the Flow setting to compensate for this. Therefore your current approach is *the wrong tool for the job*. You are asking for a feature which is already implemented, just using a different setting name and different method of communicating it through the g-code.

If you'd like to prove me wrong, please answer the following questions:

1. Does your feeder motor change step sizes depending on the filament?
2. Does your feeder gear ratio change depending on the filament?

Shine753, 2021-06-16:
The Steps/mm parameter says, as far as I'm not wrong : how many steps do I have to do to push/pull 1mm of filament. Isn't it ?

On that assumption :

> I currently think that your real problem is slip

Yes, and in my opinion, the correct answer is what is asked for, for part.
The same gear does not slip identically depending on the filament it is grabbing.

More, this is only 50% of the problem. The remaining 50% depends on the temperature you are extruding.
To extrude 1mm of some filament, that slips on the gear, you don't have to push the same way at 200°C than at 220°C.

>  _the wrong tool for the job_. You are asking for a feature which is already implemented, just using a different setting name and different method of communicating it through the g-code.

Not sure, but as far as I've seen, the extrusion ration acts when filament is pushed and is also related in some kind to the extrusion width, but not when retractation happens, where the steps/mm are involved. So changing the extrusion multiplier introduces problems when frequent and important retractations are done.

I've clogged many times my extruder using the only extrusion compensation parameter. Clogs gone when I dealed with the Steps/mm AND extrusion multiplier, the first being more reliable for pushing/pulling the filament through the teflon tube, the second when pushing it through the nozzle end.

And then, the solution would be to tell the printer, that with THIS filament, using THIS temperature, to push/pull 1mm of filament, you have to do THIS number of steps.
 
> If you'd like to prove me wrong, please answer the following questions:
> 
>     1. Does your feeder motor change step sizes depending on the filament?
That's not the subject, as tried to explain before.

>     2. Does your feeder gear ratio change depending on the filament?
That's not the subject, as tried to explain before.

Shine753, 2021-06-18:
As an addition, if you ask the printer to extrude / retract 100mm of filament by gcode command, or by LCD screen, without printing anything, it should extrude 100mm of filament. Whatever the filament slips or not. The flow rate does not act here. The M92 command and/or firmware steps/mm configuration does.

closed 2021-06-21 by no-response[bot]

Ghostkeeper, 2021-06-24:
> The Steps/mm parameter says, as far as I'm not wrong : how many steps do I have to do to push/pull 1mm of filament. Isn't it ?

Indeed it does seem to say that in the description! I'll clarify that a bit.

Ghostkeeper, 2021-06-24:
> Not sure, but as far as I've seen, the extrusion ration acts when filament is pushed and is also related in some kind to the extrusion width, but not when retractation happens, where the steps/mm [...]

This is true, but the slip ratio is also different for retractions and unretractions from the normal extrusion, since those move the filament much faster than extrusion. So having a single factor that works on both extrusion and retraction/unretraction is not a solution to your problem. Adjusting the retraction distance and flow rate to compensate for the slip is a solution.