Ultimaker/Cura #9925: Infill Line Multiplier missing from print settings - macOS

davidewen, 2021-05-30:
### Application Version

4.9.1

### Platform

macOS

### Printer

Prusa

### Reproduction steps

Looking for Infill Line Multiplier in Print Settings, it's not there.

Look in Preferences, it is there and checked.
[image]

### Actual results

Still not available

### Expected results

I expect to be able to use the option

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

no addition info

yet-another-average-joe, 2021-05-31:
Maybe you uploaded the wrong screen capture ?
Infill Line Multiplier is under Infill, not Quality.
Type "multiplier" in the search field.

davidewen, 2021-05-31:
> Maybe you uploaded the wrong screen capture ?
> Infill Line Multiplier is under Infill, not Quality.
> Type "multiplier" in the search field.

You are correct. I realized that after I uploaded the screen shot.

I also discovered that Infill Line Multiplier is only an option for a couple of the infill patterns. I had chosen a pattern that was not available. Once I selected a pattern that the option is available for, it showed up.

closed 2021-05-31 by davidewen