Ultimaker/Cura #9926: OpenGL

bttguy, 2021-05-30:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Download some of the latest releases (4.8 or 4.9)
Try to open

### Actual results

[image]
I get this error when I try to launch Cura. In the previous versions I had no issue. Nothing has changed on my laptop, all drivers are up to date.

### Expected results

Can run Cura

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fvrmr, 2021-05-31:
Hi @bttguy thank you for your report.
It is not a issue with Cura but with unfortunate with your pc.
Here you can find a duplicate https://github.com/Ultimaker/Cura/issues/5474 because it is a duplicate I will close this issue.

closed 2021-05-31 by fvrmr

bttguy, 2021-05-31:
Thanks @fvrmr So there is nothing to do? A few release back Cura worked, but now I get this error? So if it was working and my laptop could run Cura, now why could it decide to stop running it? I used Cura for almost a year without issue.

fvrmr, 2021-06-03:
Hi @bttguy the only thing you could try is to update your graphic drivers.
Cura needs OpenGL to run, so other than updating your graphic drivers there is unfortunately nothing I can help you with.

Ghostkeeper, 2021-06-08:
Nothing in this regard changed since Cura 3.3 (when the 3D layer view geometry was introduced). It's possible that the graphics drivers on the computer changed though.