Ultimaker/Cura #9927: Monitor in portrait.

coldsilk, 2021-05-30:
### Application Version

4.9.1 (AppImage)

### Platform

Ubuntu 20 (KDE)

### Printer

Custom

### Reproduction steps

Opened Cura with monitor in portrait.

### Actual results

Sizing appears to be for landscape

### Expected results

Portrait sizing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is a guess, but it seems as if Cura is always using the vertical height of landscape as the height no matter the orientation.

If I rotate my monitor and set it back to landscape everything works fine. If I rotate it back to portrait and set the manager to portrait it appears as if the entire left side of the screen is cropped off, seemingly by the amount which would be used for landscape.

I do have certain scaling features of KDE enabled, however all scaling is seemingly working correctly in Cura no matter orientation, it's only this weird crop factor that is the problem.

fvrmr, 2021-05-31:
Hi @coldsilk thank you for your report.
For Cura works in portrait mode on Windows. I will bring it up for discussing with the team. 
Keep you posted!

fvrmr, 2021-06-01:
Hi @coldsilk I have discussed it with the team and the reason why this is happening is because of the minimum width we are using for Cura. And we don't want to change, because it would be difficult to make Cura responsive, so I will close this issue.
Hope you understand. 

The reason why it was working for me is because my monitor is big enough to fit Cura in portrait mode.

closed 2021-06-01 by fvrmr