Ultimaker/Cura #9931: Finding USB printer takes too long and it's unclear if it is still working on it.

mspoorendonk, 2021-06-01:
### Application Version

4.9

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

1. Plug in the USB cable
2. Turn on the printer
3. Start cura
4. Slice
5. Wait sometimes minutes before the 'print to USB' appears in the button.

### Actual results

Cura is scanning all the baudrates and this takes a long time. I also have the feeling that it sometimes crashes the printer, but I can't confirm that for sure.

### Expected results

I see two solutions:
1. Make the USB scanning process visible by showing progress or a status message somewhere. That way you know what you are waiting for.
2. Make cura remember the last working com port and baudrate. Let it first test that. Then try different com ports with the same baudrate and only as last resort start scanning the whole bunch. Any error on opening the com port should skip it completely and proceed to the next port.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6579936/cura.log)

fvrmr, 2021-06-03:
Hi @mspoorendonk thank you for your report. 
I would see this issue more as a feature request than bug report, since you have 2 solutions for this. 
When the USB connection is successful Marlin restarts the printer, so that could be the case instead of crashing. 

On the solutions you are suggesting, I see value but Ultimaker printers doesn't use USB printing. So in that case I will defer this issue, because it is not relevant for our printers. 

Here you can find a other feature request about manual connecting to printers with USB printing: https://github.com/Ultimaker/Cura/issues/5207 that could be related to your second solution.

That being said a pull request for this is always welcome.

mydevpeeps, 2021-10-12:
@fvrmr  I would like to work on this. Can you assign this to me?

Ghostkeeper, 2021-10-19:
I can certainly assign someone.

Regarding a tag for USB printing issues, that's possible too but it doesn't quite fit in our current categorisation. Could get quite messy if this leads to the logical conclusion of also having tags for support, tags for infill, tags for shader support, etc. I will happily leave that choice up to @fvrmr who is our technical community manager and quite good at deciding how to organise these things.

closed 2023-07-13 by github-actions[bot]