Ultimaker/Cura #9933: Problems with thin objects

Prorok13, 2021-06-02:
### Application Version

4.9.1

### Platform

win10

### Printer

Uni3D

### Reproduction steps

cylindrical objects with a wall thickness of one line(0.4 mm) are processed incorrectly

### Actual results

in some places, the trajectory looks like straight lines rather than a circle.

### Expected results

the trajectories should look like circles

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

cylindrical objects with a wall thickness of one line(0.4 mm) are processed incorrectly, and if the wall thickness is 2 lines (0.8 mm), everything is normal. In version 4.8, everything is fine.
screenshot (https://user-images.githubusercontent.com/85233959/120481397-bef41c80-c3b8-11eb-8145-3164c338df4a.jpg)

nallath, 2021-06-02:
Could you also upload a project file? The screenshots you posted only show the translated setting names, so it's a bit hard for us to figure out what settings you changed.

What i can see from the screenshot is that for some reason it's only doing the gap filling (the walls are yellow and not red).

Furthermore; Thin objects are always hard. You might want to check out the arachne version of Cura. It has a brand new algorithm to handle cases like this.

Prorok13, 2021-06-03:
Project file in the attachment. 
[Prostavka.zip](https://github.com/Ultimaker/Cura/files/6591167/Prostavka.zip)

fvrmr, 2021-06-04:
Hi @Prorok13 thank you for your project file.

I could also reproduce this in 4.8 but looking at your models they are not the same, is that correct?

Also when slicing in Arachne it works fine, see here:
[image]

You can check out the Arachne version here: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta

Ghostkeeper, 2021-06-14:
Your model has a thickness of 0.499mm there: 
[image]

You have set the Horizontal Expansion setting to -0.086mm, so it'll detract 0.172mm total from that width (0.086mm from both sides of the wall). So it's supposed to print a wall with a width of 0.327mm. This is less than the Outer Wall Line Width, so it is not expected to print this with the outer wall. It is expected to print this with the gap filling as per the "Print Thin Walls" setting.

The gap filling technique has problems of its own. That makes this issue a duplicate of https://github.com/Ultimaker/Cura/issues/4685.

closed 2021-06-14 by Ghostkeeper