Ultimaker/Cura #9934: (4.9.1) Cura intermittently injecting nozzle temperature changes

PouncySilverkitten, 2021-06-02:
### Application Version

4.9.1

### Platform

MacOS 11.2.3 M1

### Printer

Custom

### Reproduction steps

Configure a custom printer as follows - 
[image] (no custom extruder start or end code, nozzle offsets 0.0mm, nozzle size 0.4mm, compatible filament diameter 1.75mm).

Set extruder 1 enabled, printing using custom material (attached, but with .txt appended to the file name so Github accepts it.). Disable extruder 2.

Slice STL (I've observed the bug occurring only intermittently, but a common factor is that the models all have a z size greater than about 50mm; that may be coincidence.)

Save file locally, transfer to MicroSD card, print.

### Actual results

Hotend temperature commands are issued to set the temperature as low as 175 Celsius during the course of the print.

### Expected results

Hotend temperature should remain stable at 215 Celsius per profile settings.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6584717/cura.log)

[custom_material.xml.fdm_material.txt](https://github.com/Ultimaker/Cura/files/6584731/custom_material.xml.fdm_material.txt)

[project.3mf.zip](https://github.com/Ultimaker/Cura/files/6584739/project.3mf.zip)

[case.gcode.txt](https://github.com/Ultimaker/Cura/files/6584747/case.gcode.txt)

GregValiant, 2021-06-02:
In the gcode file both extruders are enabled.  In the project file extruder 2 is disabled.  But when I first sliced the file, extruder 2 showed up in the gcode and was used for the support.  The "Generic" PLA of extruder 2 has a "Standby Temp" of 175 and that's what you are seeing.
Somebody will check the log but if an extruder is disabled, it should be disabled.  Toggling the "disable" checkbox did result in Extruder 2 being disabled for a subsequent slicing.  One thing about your project is that there is a single part and extruder 2 is being used for the support.  This might involve Cura's creation of the 3mf file(?).  Maybe something went sideways there.  So I agree there is a bug in there someplace, but I don't think it's about temperatures.  It seems to have to do with enabling/disabling an extruder that is designated for support.
This message came up a couple of times.
**"Settings Updated"** "Settings have been changed to match the current availability of extruders: [support_extruders_nr]

fvrmr, 2021-06-03:
Hi @PouncySilverkitten thank you for your report.
When I enabled the second extruder and changed the support extruder to extruder 1, heating up the extruder 2 isn't in my gcode. 
And then extruder 1 stays on 215 degrees. 
The 175 you see in the gcode is like @GregValiant mentioned the standby temperature. 

Do you need extruder 2 for your support? Or you don't want to use it at all?

Ghostkeeper, 2021-06-10:
Since extruder 2 is disabled, it's expected that it won't get used. The front-end also adjusts its disallowed areas based on that. Because those disallowed areas are wrong, you could place something that gets printed outside of the build volume.

The reproduction steps in this ticket don't reproduce it. In the project file I found that the custom profile overrides the support extruder to be the second extruder, which is disabled. Based on that, I found the following reproduction steps for this bug:

1. Add a new dual-extruder printer.
2. Enable support.
3. Change the support extruder to the second extruder.
4. Save the current changes to a new custom profile.
5. Change to one of the default profiles.
6. Disable the second extruder.
7. Change back to your custom profile.
8. Slice something that would generate support and inspect the g-code.

evtrados, 2021-09-22:
Ooops, my apologies! I was assigned to try and reproduce it on my macbook and then report back, but it seems I never did the latter...
I re-checked again today just to be certain and indeed I also could not reproduce the issue. That being said, I don't think this issue would be OS-specific, since as it is also mentioned above, the fact that extruder 2 is disabled makes it expected that it will not get used.

closed 2024-12-09 by github-actions[bot]