Ultimaker/Cura #9935: Support Generation Using Infill Steps

jlow6978, 2021-06-03:
### Application Version

4.9.1

### Platform

Windows 10 X64

### Printer

Any, not relevant to the issue.

### Reproduction steps

Set up for slicing using 3 infill steps for a support
Using GRID Support Line Distance 4 Support Infill Step Height 3mm
The part I am making is round at each layer.
it is a shallow cone shape so there is support everywhere.

### Actual results

As the layers increase, at the step, material is extruded into the air with no chance of staying in the expected position.
So a bridge can occur only when the new layer can fuse to material on the previous layer.  
My screen shot shows a line segment ending in mid air followed by a travel to another segment starting in mid air.
This is wasted material that can damage the print.  see lower right of image above save to box.
[image]

### Expected results

At a new infill step the new segment should not extend past the prior step box unless it can bridge to the other side of the prior step box.

[CFFFP_Spiral1088a.zip](https://github.com/Ultimaker/Cura/files/6593377/CFFFP_Spiral1088a.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

I wanted to use tree supports but could get not get a set of parameters that would work.

fvrmr, 2021-06-04:
Hi @jlow6978 thank you for your bug report. 
This is kind of expected behaviour, but I understand that you could see this as wasted material. But the layers will be build up and will get to a point to be bridged. You could increase the gradual support infill steps. 
I made a image to make it clear:
[image]

This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/9374 because the gradual infill steps works the same as support infill steps, so I will close this is as a duplicate.

But let me know if you have more questions in the other tread. So we keep everything in one place.

jlow6978, 2021-06-04:
Thank you for your response.
However, I disagree with you decision to Close this issue.
This is a very smart methodology for infill and supports that saves a lot of time and material.
While I got a successful print using this method you can see the amount of failed bridging that actually occurred in my print in the attached photo.
[image]

The fact tat you had someone else raise a similar issue suggests that this methodology can and should be improved.

I know that I could write a routine to post process the G Code file to eliminate line segments that are not bridges.  
This is based only on the previous layer and a good segment would had two points of contact with the previous layer.
If the segment did not have the ability to bridge (two points of contact) it would be eliminated.
So at any step you are only looking at the corrected previous layer and reapplying the process.

Maybe you have higher priority tasks to complete but I think it would be a mistake to not make this a task for the developers.
BTW, when I was trying to make the Tree Support work for my project, I also saw this same behavior when a segment would be printed in mid air.
So my suggested post processing routine could be applied there also.

Ghostkeeper, 2021-06-11:
That doesn't take away that this is a duplicate. Your remarks would be better added to the ongoing discussion at #9374. Here they will get lost. We do listen to arguments like those, but if every commenter starts a new thread about it, that gets very confusing. This was correctly marked as a duplicate.

closed 2021-06-11 by Ghostkeeper