Ultimaker/Cura #9936: (4.9.1) Cura does not load my printer profile

Zhern, 2021-06-03:
### Application Version

4.9.1

### Platform

Arch Linux

### Printer

Sovol SV01

### Reproduction steps

1. I launch "cura" a very first time, saved a printer profile, some materials, custom profiles, and then close.
2. I launch "cura" some time later.

### Actual results

Sometimes cura launches perfectly fine and I start right into the application without any splash screen, I have my profiles loaded, etc.
Sometimes cura launches quite wrong and I start into the Welcome splash screen asking me to create a new printer profile, not showing the one I already created before.
This is totally random, I would say I have 50% chances to get it to properly launch. I can have 10 failed launches in a row, as well as 10 successful launches in a row.
I tagged this to the 4.9.1 version because this is the one I currently have, but I know this bug is there for a long time now, I would say at least from 4.7.

### Expected results

I expect the application to properly launch everytime, without me having to launch N times the application until it loads everything.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I did some digging, and I ended up in the file `UM/Settings/ContainerRegistry.py`, function `findContainersMetadata`, seeing that the metadata are not always outputting the same thing.
In the [logs](https://github.com/Ultimaker/Cura/files/6593860/cura.log) I can see at line 401:
`2021-06-03 21:16:42,703 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition SV01 failed to pickle.`
This message comes from Uranium if I understood well.
I could not figure out what the root cause is, maybe from cura, maybe from uranium, maybe from another dependency/library.

fvrmr, 2021-06-04:
Hi @Zhern thank you for your report.  I can see that you are using phyton 3.9 are Cura runs on 3.8 so that might causing it.
Did you try to run the AppImage?

Zhern, 2021-06-05:
I just tried to run the appImage 4.9.1 twice, the 1st time was fine, the 2nd was not (Welcome splash screen and no loaded profile).

fvrmr, 2021-06-07:
@Zhern can you share your log file of that as well?

fvrmr, 2021-06-07:
Thank you. I added your 3D printer and I couldn't reproduce it out of 20 times trying. 
I couldn't find anything else in your log file, other than that this line again:
`2021-06-07 10:50:09,156 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition SV01 failed to pickle.`

Could you share your config folder with me?

Zhern, 2021-06-07:
Sure, [here](https://github.com/Ultimaker/Cura/files/6611131/config.zip) are the `~/.config/cura/4.9` and `~/.local/share/cura/4.9` folders.

fvrmr, 2021-06-08:
Thank you, I can reproduce this now! I will bring it up for discussion with the team. Keep you posted!

nallath, 2021-06-08:
Hmm. Could you try to delete your ~/.cache/cura/4.9 folder?

Zhern, 2021-06-08:
I moved the whole ~/.cache/cura/ folder, the 1st run showed the Welcome splash, the 2nd and 3rd did not, the 4th did it again. :(

nallath, 2021-06-08:
The cache is the place where we store the pickled files, which is where the "The definition cache for definition SV01 failed to pickle." issue might come from. 

Since we create the cache during the first start, i hoped that that would explain why the second & third starts would fail (since at that point you're loading the cache and not the "raw" file). But that doesn't really seem to be the case :(

Ghostkeeper, 2021-06-14:
Seems to be random then, maybe?

closed 2024-12-09 by github-actions[bot]