Ultimaker/Cura #9938: OS X can´t click

Raven420, 2021-06-04:
### Application Version

4.9.1

### Platform

macOS Big Sur 11.4

### Printer

Anycubic I3 Mega

### Reproduction steps

1. Load a file
2. I tried to rotate it but I couldnt klick the thing
3. I selected the object with the object list but I still can´t click it 
4. You can´t use the arrows aswell. Its like you can´t click any object.

### Actual results

nothing happens

### Expected results

you should be able to move the object with the 3 arrows or rotate it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It worked on the version 4.8.0

fvrmr, 2021-06-04:
Hi @ravetsch thank you for your bug report.
It looks like this is a duplicate of https://github.com/Ultimaker/Cura/issues/9912 do you also have a second monitor?

If that is the case it is a duplicate of the thread above, if not could you also share your log file?
You can find your log file here: `$USER/Library/Application Support/cura/<Cura version>/cura.log`

jtusman, 2021-06-07:
I have the same problem on a Mac running OS X 10.15.4 Catalina.
Cura 9.0 cannot click on or move any files.
Removed Cura 9.0 and installed Cura 9.1.
It worked once and then stopped working as before.
I have attached the log file to help find the problem.
[cura.log](https://github.com/Ultimaker/Cura/files/6611366/cura.log)

fvrmr, 2021-06-11:
@jtusman do you have a second monitor?

jtusman, 2021-06-11:
Yes I do have 2 monitors.

jtusman, 2021-06-12:
I'm using a MacBook Pro with 2 additional monitors.

Ghostkeeper, 2021-06-14:
Try moving the Cura window to the MacBook's own screen, which is reported to be a workaround here: #9912.

We'll close this as a duplicate to consolidate the discussion in one thread.

closed 2021-06-14 by Ghostkeeper

jtusman, 2021-06-15:
I moved the Cura 4.9.1 window to my MacBook and it does work.
But the screen is small and not my first choice to use.
Is there a fix in the works?

Ghostkeeper, 2021-06-21:
@jtusman This is a duplicate ticket. Please focus the discussion in the original. Nobody wants to gather up all information from all of the duplicates.

jtusman, 2021-06-21:
This is why I posted where I did.

*Joseph Williams*

Jun 16, 2021, 21:21 GMT+2

Hi, Jack,

It looks like your case was closed and merged into an ongoing ticket with a
user that is also having the same issue. Since they're in the process of
solving the problem, I recommend adding any additional information you have
to the link below.

https://github.com/Ultimaker/Cura/issues/9912

Joseph Williams

Ultimaker North America Support

For any questions, comments or concerns feel free to consult our support
page

www.UltimakerNAsupport.com