Ultimaker/Cura #9939: Cura 4.9.1 stuck on Initializing Build Volume

clarkot, 2021-06-04:
### Application Version

4.9.1

### Platform

windows 10

### Printer

Ender 3 v2

### Reproduction steps

1. I have attempted to restart the program, restart my computer, ensuring that my computer is updated. This has worked to load Cura once or twice. 
2. The alternative that has worked with some more consistency is that I uninstall and reinstall the program. The downside to this is I have to reapply my settings and profiles which after doing this a dozen times it gets old. 

### Actual results

1. Restarting the computer would only work a few times. 
2. uninstalling and reinstalling the program worked with more consistency. Although I would have to rebuild all of my profiles and settings which just gets old after doing it so many times. 

### Expected results

Like I stated previously, restarting the computer would work occasionally, but uninstalling and reinstalling would work more consistently.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

worldpoop, 2021-06-04:
Not just Windows 10.  Same happening on Mac (High Sierra) right now.  Stuck at "Initializing build volume" :(

clarkot, 2021-06-04:
> Not just Windows 10. Same happening on Mac (High Sierra) right now. Stuck at "Initializing build volume" :(

Very frustrating! I hope they can fix it

billyjacoby, 2021-06-06:
I'm experiencing the same issue on Big Sur using an M1 MacBook.

fvrmr, 2021-06-07:
Hi @clarkot thank you for your bug report. This is a duplicate of https://github.com/Ultimaker/Cura/issues/9847
This is fixed for the next release. 
I will close this issue because it is a duplicate.

closed 2021-06-07 by fvrmr