Ultimaker/Cura #9940: Crash on start up

matty2233, 2021-06-05:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ultimaker Original

### Reproduction steps

Started program
After its Small set up window program looks to open and then just crashes.

### Actual results

crash

### Expected results

open cura

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nope

fvrmr, 2021-06-07:
Hi @matty2233 thank you for your report.
Could you also share your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

It could alsobe a duplicate of https://github.com/Ultimaker/Cura/issues/9847

Mareckil, 2021-06-08:
I have the same issue, Graphic card Nvidia Quadro (two different models K2200 and P620) latest drivers but with older it was working same way. Two monitors as extended screen.
Problem is when program start on second monitor, while it runs on first it works fine. In my case I have zero (datum point) in the center of table and as long this point stays on first monitor all is working fine but when it pass the edge it freeze on start.
Work around is to uncheck restore windows position on start in config or delete window left and window top coordinates from config file

[cura.log](https://github.com/Ultimaker/Cura/files/6615156/cura.log)

nallath, 2021-06-08:
@Mareckil I have a potential solution for this, but i've not been able to test it (as I don't have the issue myself). See https://github.com/Ultimaker/Uranium/pull/705

Mareckil, 2021-06-08:
thanks @nallath, unfortunately I am not programmer, I just have a problem with cura and was searching for solution. I did some tests and investigation. I did find this thread so shared my observations so might some one like you can resolve this issue, because as I can see it is recurring problem.

nallath, 2021-06-08:
It's a problem that is recurring, but as you might have already been able to read from other issues, it's also one that we haven't been able to resolve for quite some time now.
The biggest issue that we have is that non of our developers have the issue themselves, which makes it really hard to debug it. The fix i made is a theoretical fix. With a little luck it will end up in the next Cura release (4.10)

Mareckil, 2021-06-08:
Well, if I can help anyhow, might providing more detailed debug file or test new fixed program on my affected platform, just let me know. Only thing I have to get compiled files.

chiefmaster81, 2021-06-11:
[stderr.log](https://github.com/Ultimaker/Cura/files/6637559/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/6637563/cura.log)

evtrados, 2021-06-11:
@Mareckil ,

Are you perhaps using a docking station for your external monitor(s)? Or is it directly cable-connected to your graphics card?
I've seen issues where Cura would crash when opening on an external monitor which is connected to a docking station. Your issue, however, reminds me of one I had encountered quite a while ago, where the Cura window would appear for a glimpse of a second and then it would just crash without any message. That issue was due to the Intel GPU failing and would be resolved by removing the graphic drivers.

I tried reproducing your issue on 2 machines but I could not unfortunately. The fix by @nallath will be in the 4.10 release, so please check on that version if this fixes it for you.

chiefmaster81, 2021-06-11:
no, no dockin station. 4 dirctly connected displays. 2 via Displayport and 2 via HDMI .

fvrmr, 2021-06-11:
@chiefmaster81 could you try running Cura without the external monitors?

Mareckil, 2021-06-12:
@evtrados My monitors are connected directly to graphic card trough display ports, In my case it began when I changed mainboard, memory and CPU. Graphic card was the same as in old computer. Of course I had to install fresh Windows 10 Home 64bit. On old computer it was working on new it freeze.

chiefmaster81, 2021-06-12:
> 
> 
> @chiefmaster81 could you try running Cura without the external monitors?

no change with only one display. issue persists. I tried both, HDMI and DP .

Mareckil, 2021-06-13:
Good new. So far. after recent windows update cura opens up properly on second screen, at least on my computer.

chiefmaster81, 2021-06-14:
Update did not fix it for me. still crashing.

Ghostkeeper, 2021-06-14:
> Update did not fix it for me. still crashing.

Which update did you try?

Ghostkeeper, 2021-06-14:
This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/7183.

closed 2021-06-14 by Ghostkeeper

chiefmaster81, 2021-06-14:
KB5004476

Mareckil, 2021-06-14:
Well I am back, I t seem strange, but today it doesn't start again, nothing changed. Just turned computer off yesterday, turned it on today and it's back :(. It looks like something on my computer works randomly, I do have two programs having problems: cura while it start on second screen and Mastercam when closing application. Both programs works fine Cura when it is on first display and mastercam until I would like to close it, it hangs up, so I have to kill process. Yesterday both programs where working properly! 
In task manager, waiting chain analysis for cura.exe shows: at least one thread of Cura.exe process waiting to end network operations in/out
And for Mastercam: At least one thread of Mastercam process is in waiting stage.

fvrmr, 2021-06-21:
Cura 4.10 beta is released last Friday and we had a possible fix for this issue. Please try if Cura 4.10 beta doesn't crash on start up. 
https://github.com/Ultimaker/Cura/releases/tag/4.10-beta

chiefmaster81, 2021-06-21:
great ! It works again for me. Cura is starting ! Thank you !

matty2233, 2021-06-27:
Yes this has been solved for me in the latest 4.10 update.