Ultimaker/Cura #9941: Cura 4.9.1 fills gaps between walls after ironing

Grumpher, 2021-06-05:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

CR6-SE

### Reproduction steps

Slice an object with:
Ironing top layer enabled 
Fill gaps between walls enabled

### Actual results

Printer prints top layer as expected, then irons the top layer with a good surface finish. then the printer ruins the finish by "filling" the small gaps that no longer exist after ironing.

Could be fixed by filling before ironing, or allowing an option to fill small gaps for every layer except the top surface

### Expected results

Expected to iron as the very last operation, leaving a good surface finish

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Sorry if the photos aren't the best, this is the finished result after some small cleaning with flush cuts. You can see where there are no small gaps the top surface is perfect. then where small wall gaps are present the surface is pushed up in the final stage of the print

Fill small gaps is needed elsewhere in the print except for the ironed layer

[image]
[image]
[image]
[image]
[image]

fvrmr, 2021-06-07:
Hi @Grumpher thank you for your report.
Could you share your project file? Than I can see if I can reproduce this is Cura.
Thank you in advance.

Grumpher, 2021-06-07:
> 
> 
> Hi @Grumpher thank you for your report.
> Could you share your project file? Than I can see if I can reproduce this is Cura.
> Thank you in advance.

Project file attached with geometry
[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/6606984/New.Compressed.zipped.Folder.zip)

fvrmr, 2021-06-08:
Thank you 
Looking into the preview tab it is ironing that will cause this. Look at the thin lines. 

[image]

Did you try to change the infill pattern to zigzag or decrease the ironing flow?

Grumpher, 2021-06-08:
> 
> 
> Thank you
> Looking into the preview tab it is ironing that will cause this. Look at the thin lines.
> 
> [image]
> 
> Did you try to change the infill pattern to zigzag or decrease the ironing flow?

@fvrmr 

I'm aware ironing will cause the issue to present, that is the source of the issue post. Or more specifically the combination of ironing and Fill gaps between walls

I haven't tried different top patterns or lower iron flow, I don't see how it will help as the issue stems from the fact that ironing is not the last operation when Fill gaps between walls is turned on

it's not a huge issue but when ironing is turned on it should always be the last operation, because that is the point of ironing?

Ghostkeeper, 2021-06-14:
> Or more specifically the combination of ironing and Fill gaps between walls

But your print does not fill gaps between walls:

[image]

And indeed it is leaving a gap between the walls here, which would otherwise get filled:

[image]

As far as I can tell your problem stems from that it starts with these contours, belonging to the bigger part:
[image]

Once it is done with those it still needs to fill the remaining inner parts:
[image]

The order here stems from that it prints the ironing from the inside out, starting with the innermost contour, which is all the way on the left side of the print somewhere. It continues that contour all the way to the outside, to provide the fewest possible interruptions. As it expands towards the outer wall, the contour grows to include the entire shape, encompassing also other parts that haven't been filled yet. It then fills those afterwards.

I don't really see how Cura could fill those up better, then. What would you suggest as expected behaviour?

Grumpher, 2021-06-14:
@Ghostkeeper 
Sorry the project I sent must have been after I tried fill gaps off. The issue occurs exclusively with fill gaps on.

The iron stage works fine, as expected, there is a perfectly smooth surface on the top layer, and then a secondary process comes in and re does where the walls gaps exist. I watched it happen on the printer as it occurred.

Again I'll repeat to make myself clear, Ironing without fill gaps turn on produces great top layer, fill gaps between walls works fine on its own. When combined the fill gaps between walls takes priority over being last operation and so happens after ironing

Expected behavior is to iron last, after filling gaps between walls, doing this would produce a perfectly smooth top layer as expected

Ghostkeeper, 2021-06-14:
Ah, but then I can't reproduce your issue. These are my reproduction steps:

1. Load `Hex Holder - Part 2.3mf` as a project file.
2. Change the Fill Gaps Between Walls setting to "Everywhere".
3. Let it slice and go through the last layer with the horizontal slider.

This is what I'm seeing. First it prints the walls and skin:

[image]

Then it prints the gap filling:
[image]

And finally it prints the ironing:
[image]

Are you seeing a different ordering?

closed 2021-06-28 by no-response[bot]