Ultimaker/Cura #9942: Cura 4.9.1 Bottom layers printing twice as wide as specified when pattern set to Line.

IMakeToo, 2021-06-05:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Custom (Taz6 with E3D Volcano printhead & Moarstruder firmware; 1.2mm nozzle)

### Reproduction steps

I created a simple tray in Fusion 360 with a shell thickness of 2.4mm, (2 x the nozzle diameter), then sliced in Cura with a layer height of 0.8mm and line width of 1,2mm (same as the nozzle diameter). When printed with the Top/Bottom Layer Pattern set to Lines or ZigZag, the pattern lines on the 3 bottom layers all print 2.4mm wide and 2.4mm apart, twice the setting in Cura. They print at twice the width and spacing specified, no matter what size I specify for the width.

The walls print correctly, and the preview in Cura appears correct.

If I change the Top/Bottom layer pattern to Concentric the bottom layer lines all print at the correct width and spacing.

### Actual results

Image of the bottom printed with Lines pattern and Concentric pattern, showing incorrect spacing and width.
[image]

### Expected results

The lines for the bottom layers should all print at the width and spacing setting in Cura, not twice the size specified.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file.
[BottomSkinIssue.zip](https://github.com/Ultimaker/Cura/files/6602267/BottomSkinIssue.zip)

smartavionics, 2021-06-05:
Looks very over-extruded to me. A quick test would be to reduce the flow for the skin to, say, 50% and that would let you check the line spacing.

IMakeToo, 2021-06-06:
Thanks for that suggestion. Reducing the top/bottom flow rate to around 67% gave expected results for the initial layer, but then the subsequent bottom layers were under extruded. Reducing just the initial layer flow rate to 50% produced a reasonable initial layer, but then the following top/bottom layers were over-extruded. Testing with reduced flow rates for both the initial layer and the top/bottom layers was time-consuming because they seem to interact. I got reasonable results at 80% for the initial layer and 67% for the top/bottom layers.

Note that I have only needed to change these flow settings when using line or zigzag top/bottom patterns. Concentric works as expected with default (100%) flow settings. Also I did not notice this issue using the stock Taz6 printhead with 0.5mm nozzle and standard firmware. Could it be somehow related to the large nozzle size and/or Moarstruder firmware?

I have attached a smaller project that prints more or less as expected with the 80%/67% flow settings.

[BottomSkinReducedFlow.zip](https://github.com/Ultimaker/Cura/files/6603488/BottomSkinReducedFlow.zip)

smartavionics, 2021-06-06:
>  I got reasonable results at 80% for the initial layer and 67% for the top/bottom layers.

That suggests to me that the nozzle is too far from the build plate for the first layer.

> Could it be somehow related to the large nozzle size and/or Moarstruder firmware?

You have calibrated the extruder steps, haven't you?

IMakeToo, 2021-06-06:
Yes, I have calibrated the extruder steps, and spent a fair bit of time checking that the nozzle is the correct distance from the build plate. When I set the layer height to 0.8mm, the walls and skirt measure 0.8mm within 0.05mm. The build plate is fairly new, and is flat within about 0.1mm.

 What I can't understand is why the walls and skirt appear to print correctly, but the skin doesn't not unless I choose concentric. If the nozzle height or eSteps were incorrect, then surely the walls and skirt would not print correctly, and I would not see expected results when I use the concentric pattern for the bottom layers.

It seems as if somehow extrusion is not being calculated correctly for the bottom layers in Cura in some cases, and maybe only when using larger nozzle sizes.

When I get some more time I will try the Lulzbot version of Cura, and maybe PrusaSlicer to see if I have the same problem.

fvrmr, 2021-06-07:
Did you also try to print this with previous versions of Cura, could you try 4.8 or 4.7?

IMakeToo, 2021-06-08:
I have just tried with the Lulzbot edition of Cura (3.6.23) and experience the same problem. I really can't figure this one out. I have enabled all settings in the profiles and experimented changing anything that looks remotely like it could be causing this, but nothing makes any difference except altering the flow rates as mentioned above.

smartavionics, 2021-06-08:
I know you said you calibrated the extruder but perhaps it's worth checking again as an incorrect extruder steps/mm value is an obvious cause for over (or under) extrusion.

IMakeToo, 2021-06-08:
Okay I’m fairly sure I have found the cause of this issue, and it’s not Cura. Zach from IT-Works3D suggested I look at the Z-offset again which I did and found that reducing it (i.e. setting the printhead further away from the bed for the first layer) overcome the issue completely. I probably should have tested that in the first place, but I was misled by the way the issue only seemed to show up on the skin layers and not the wall or skirt, and not when using the concentric pattern.

closed 2021-06-08 by IMakeToo

fvrmr, 2021-06-08:
Hi @IMakeToo thank you for letting us know!