Ultimaker/Cura #9943: cura -craftbot interaction: incomplete print

damidb, 2021-06-05:
### Application Version

4.9.1

### Platform

windows 8.1

### Printer

own profile (for Craftbot plus printer)

### Reproduction steps

I mad a stl file with autodesk fusion.
I made ultimaker Cura slice the stl file, because it's been said to be the best slicer.
Cura reported: 4h and 4.5m printing time/filament use. Looked fine to me.
 

### Actual results

But craftbot , once starting to print, said 40 min, 0,6 m filament (?). I let it print it out of curiosity, off course it was only a very partial print (uncomplete object)  when it stopped after the 40 min.
Craftbot plus seemed to me to stop this print at around the same moment I deliberately stopped a first testprint of the same object with Craftbot plus, a few days ago,
That stop was necessary because something went wrong at that time (I had to pause the print (should be no probleem with craftbot plus), but when I restarted the print fews hours later, the extruder fid hit (and moved) the first already printed part of the print).

### Expected results

I expected Craftbot to  print 
the object completely, with 4h and 4.5m printing time/filament use

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Garminholder-V3-3.5%bigger.3mf.zip](https://github.com/Ultimaker/Cura/files/6602707/Garminholder-V3-3.5.bigger.3mf.zip)

fvrmr, 2021-06-07:
Hi @damidb thank you for your report.
Could you share your gcode of this project as well? 
In the preview tab I can't see anything happening that will cause the stop.

GregValiant, 2021-06-07:
It would also be good to know if you were printing from the SD card, or using  USB Printing from Cura.

Bad sectors on a memory card is the usual culprit for stoppages when SD printing (and they are usually terminal because the gcode file is corrupt).  USB printing can also cause random stoppages but are often recoverable if you are right there and can "click to resume" the button on the LCD.

Re-starting a print is a practiced art.  Once the stepper motors have become disabled (the default is 2 minutes) then the printer requires an Auto-Home move to figure out where it is.  That can often result in a collision when the print restarts.

closed 2021-06-21 by no-response[bot]