Ultimaker/Cura #9949: Cura and screen readers (JAWS, NVDA, VoiceOver)

erbuckwalter, 2021-06-08:
### Is your feature request related to a problem?

Backstory: The agency I work for is looking at purchasing a 3D printer to make accessible items for the individuals who are blind and visually impaired. I am the one on staff with the most experience with 3D printing and so I am looking into some possibilities. I have experience with Cura and use it personally.

Of special interest to us if Cura is accessible by screen reading software (Windows: JAWS, NVDA or Apple: VoiceOver.

Also curious if the remote print features via network or iPhone is accessible with screen reader on computer or VoiceOver on the iPhone.

### Describe the solution you'd like

It would be great if Cura is accessible with screen readers, or remote print is accessible with screen readers.

### Describe alternatives you've considered

Not having a printer accessible by all staff. :(

### Affected users and/or printers

Possible purchase of Ultimaker 3 or S5

fvrmr, 2021-06-08:
Hi @erbuckwalter thank you for your feature request.
I will discuss this with the team. Keep you posted!

fvrmr, 2021-06-14:
Hi @erbuckwalter I have discussed it with the team. 
We have heard from other users that Cura works quite well with screen readers. But we haven't used it our self.
So you have to try it yourself to see if it works the way you would like it to work. Did you try to use Cura with screen readers already?

closed 2021-06-28 by no-response[bot]