Ultimaker/Cura #9951: Material->Flow Setting

adsaxton, 2021-06-08:
### Application Version

4.9.1

### Platform

MacOS 10.14.6

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Set Material->Flow to 92%
2. Slice
3. Save GCode File

### Actual results

The resulting GCode file has no adjustment to Flow (M221) except that in the Start G-code.

### Expected results

Somewhere the flow should be set as entered in the print settings.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Slicing Result.zip](https://github.com/Ultimaker/Cura/files/6620158/Slicing.Result.zip)

Asterchades, 2021-06-09:
Flow adjustments are made in the E portion of the G0/G1 movements, rather than via Marlin flow settings. I suspect it's for more universal compatibility in the event a firmware doesn't have such facilities but that's just a guess on my part.

fvrmr, 2021-06-11:
Hi @adsaxton thank you for your report.
@Asterchades is correct, the flow is adjusted in the E values. 
Setting the flow doesn't override your start gcode.

adsaxton, 2021-06-12:
It doesn't seem to effect the result one way or the other. The only way I could correct the problem was to change the start g-code for the printer.

Ghostkeeper, 2021-06-16:
To share a project file, please go to File -> Save Project. You can share the project here on Github if it's contained within a .zip archive (or renamed to .zip). The file you currently shared is a .gcode file and doesn't allow us to reproduce your issue.

closed 2021-07-01 by no-response[bot]