Ultimaker/Cura #9952: Top/Bottom -> Ironing support on top raft layer

iasj, 2021-06-09:
### Is your feature request related to a problem?

No. Just an improvement.

### Describe the solution you'd like

Hi there. 

I searched in the issues segment as `iron raft`, and no issues were found. I then suggest the creation of a ironing feature to be applied on the upper most layer of the raft surface. 

I'm assuming this could provide smoother bottom surface of the piece after it's plucked out of the raft. Also, it's possible this feature makes the removal of the raft easier, when an air gap is set too small.

Obviously, all this is just speculation, but if it's not too much trouble to develop, I think it would be nice to have this feature in the next versions.

Appreciate this project. Bye!

### Describe alternatives you've considered

no alternatives.

### Affected users and/or printers

I don't think so.

### Additional information & file uploads

There is no such feature yet.
[image]

fvrmr, 2021-06-11:
Hi @iasj thank you for your feature request. 
I had a small discussion with my colleague about this and decided to defer this. 
Because we don't think that the top part of the raft will be that smoothed with ironing, because there is gaps between it by the `raft top spacing`
So it won't add that much value to the raft. Hope you understand.

double0jimb0, 2021-09-01:
@fvrmr Slight hijack, I opened this feature request for Ironing the support interface layer (not just on raft, but any interface). Since you recently discussed with colleague, maybe it is still fresh: https://github.com/Ultimaker/Cura/issues/7012

And FYI, ironing support interfaces is standard protocol on the old uPrint SE-Plus.

closed 2023-07-13 by github-actions[bot]