Ultimaker/Cura #9953: Failed print

epicyclic, 2021-06-09:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

I sliced a couple of different Fusion 360 generated files and all appears well on the screen.  the files were loaded to printer via micro SD card.

### Actual results

The printer generates the purge line but no skirt and the printer then produces blobs prior to printing.  I checked for a clog but all ok.  
I then uninstalled 4.9.1 and reloaded it but the same happened. 

### Expected results

 Since then I've loaded 4.8.0 and sliced the same files and they have printed fine, exactly as expected, with skirt and no blobs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

See above.

fvrmr, 2021-06-10:
Hi @epicyclic thank you for your report.
Did you try to load the same project file from 4.8 in 4.9.1? Or did you create a new file?
Could you share both project files (file - save project) in a zip with me?

fvrmr, 2021-06-11:
Hi @epicyclic could you post your stl (zipped) in a comment here on Github. It doesn't send the file when replying via email.

epicyclic, 2021-06-11:
Zipped folder as requested...
[X axis adluster body v9.zip](https://github.com/Ultimaker/Cura/files/6637988/X.axis.adluster.body.v9.zip)

fvrmr, 2021-06-11:
I can't see anything strange that is causing this. 
Did you print both files (in 4.9.1 and 4.8) with the same SD card?

epicyclic, 2021-06-11:
Yes, indeed.  Everything was identical, I did try a second SD car and had the same result.
FYI, the file I copied was modified from a .step file from Thingiverse. 
I've just generated a new part to print and will try it with 4.9.1 and let you know what happens.
Thank for your help!

epicyclic, 2021-06-11:
I'm pleased to say that the new design printed fine in 4.9.1 as did a resliced version of the part that was causing problems!
I'll keep monitoring the situation and report back if I get this error again.

fvrmr, 2021-06-14:
For now I will close this issue as fixed, but feel free to reach out!

closed 2021-06-14 by fvrmr