Ultimaker/Cura #9954: 3 bugs:  Prints on Air.  Prints Backwards.  Piles objects on top of one-another

gitcnd, 2021-06-09:
### Application Version

4.9.0

### Platform

Mac

### Printer

Anycubic Chiron

### Reproduction steps

I opened 3 .stl files at once (shift click) - and they all get blobbed in a pile on top of one another.

It tries to print the inside of a bridged area on air - it needs to print the sides first, for the material to stick onto.  It does this no matter whether I select "inside out" print or not.

It behaves weird with infill skipping forward to print backwards - the code should look at what is being output, and flip directions around when the travel that gets output is obviously ridiculous.

See screencast.  https://autode.sk/3italEx

### Actual results

See here: https://autode.sk/3italEx

And: 

[image]

### Expected results

Obvious mistakes that could easily be avoided by the software probably should be...

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[demo_stl.zip](https://github.com/Ultimaker/Cura/files/6624208/demo.zip)

nallath, 2021-06-09:
Could you for the next time make seperate issue reports for each issue? That makes it a lot easier for us to handle things.

>It tries to print the inside of a bridged area on air - it needs to print the sides first, for the material to stick onto. It does this no matter whether I select "inside out" print or not. 

The part really needs to be printed with support. The current bridging routine assumes that both sides are already supported. In this case the order wouldn't matter, since both situations would lead to failure.

> It behaves weird with infill skipping forward to print backwards - the code should look at what is being output, and flip directions around when the travel that gets output is obviously ridiculous.

I tried slicing this myself, but i didn't see this issue. Could you share a project file that reproduces it? I assume it's caused by something in your configuration.

gitcnd, 2021-06-10:
"The part really needs to be printed with support. " - *irrelevant* (plus not true, but let's not discuss irrelevant things).

"The software should understand that printing on thin air is impossible" - relevant.

I understand this means someone has to wade into the guts of some really complicated software to find the bug, but that doesn't mean that this isn't a bug or that nobody should try to do that.  Regular users should not have to guess which kinds of travel moves might be made by Cura and compensate for absurd situations like thin-air-printing: the software should already understand NOT to do absurd things in the first place...  There's even a word to describe exactly this: bridging.  You never print bridges from the middle, do you?  That's stupid - because it's thin-air in the middle. You should never in general print *anything* that starts on thin air, when it *should* start from someplace with material below (or at the very least beside) it.

Also, regarding the backwards printing, I'll see if I can, but, again - it is irrelevant what my settings are - producing dozens of consecutive travel moves that, when simply each one is flipped, results in a 300% time+distance savings is probably the kind of really-simple-to-code post-processing step that you should just be adding in anyhow - regardless of what CAUSED this, it's a no-brainer to PREVENT it from happening (regardless of what other things might also cause this).  I expect there's probably already some kind of posst-proess step?  Noticing this problem suggests that's a bug in the way it considers geometry and directions of nearly-straight lines (those were dozens of tiny arcs - almost straight lines, but not quite.)

nallath, 2021-06-10:
First off; I'm trying to help you here. Please change the tone of your message. This is not the first time that I've asked you to do so, but it will be the _final_ time that I will. I am more than willing to help you fix the issues, even though you aren't a customer of Ultimaker, so the least I expect is a normal tone in your communication.

> "The software should understand that printing on thin air is impossible" - relevant.

It does, that's why it marks the areas as red. But we have opted to only warn users about certain things, as there might be cases where the software is wrong (thus allowing the software to be overridden by the user). In some cases (including overhangs) there are multiple ways to fix this, so there is no automatic solution possible. If you have an suggestions as to what behavior you would like to see, we're always open to discuss this! 

>"The part really needs to be printed with support. " - *irrelevant* (plus not true, but let's not discuss irrelevant things). 

It does need to be printed with support. The bridging can't be done completely due to the hole that is in the center. I understand what bridging is. No need to explain this to me. 

I've tried to draw out a number of options, but none of them would work. It could be that I missed one, so if you could share the strategy that you have in mind that would help a lot. 

> Also, regarding the backwards printing, I'll see if I can, but, again - it is irrelevant what my settings are - producing dozens of consecutive travel moves that, when simply each one is flipped, [...]

You might want to read what I wrote. I was unable to reproduce it, so there might be some settings that cause this effect to happen. In order to figure out what that is, i need to have a project file that reproduces this. The settings also have a complex interplay with one other, so there might be cases where certain strategies (that in this case lead to wrong results) are correct. But without seeing _what_ you did, there is little sensible that I can say, sorry.

Ghostkeeper, 2021-06-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/3513.

closed 2021-06-17 by Ghostkeeper

gitcnd, 2021-07-02:
@nallath  _"First off; I'm trying to help you here. Please change the tone of your message."_
You got that entirely backwards.  I'm wasting my time for FREE to help you fix BUGS that affect all your customers. You're not "helping me" - I use other slicer software now, because yours does not work right, but I'm still courteous enough to let you know where you stuffed up.

"It does, that's why it marks the areas as red." ... "there are multiple ways to fix this" ... it appears you fully understand both the problem and the solution(s).

"The bridging can't be done completely due to the hole that is in the center." - wrong.  "Mechanical iris" if you need a hint to help you figure that out.

Winding me up and pretending otherwise is a vastly more serious lack of respect than my "tone" that you emotionally complain about.