Ultimaker/Cura #9955: Cura 4.9.1 MacOS - gaps between walls

renestrauch, 2021-06-09:
### Application Version

4.9.1

### Platform

MacOS

### Printer

S235

### Reproduction steps

Hello

I open a Cube STL and slice, save the gcode and print. 

The prints have gaps between the walls.
Print thin walls was enabled and disabled. No change.
Line width was tested between 0.32 and 0.48. No change.
Flow was increased. No change.
Tested with Combing and Coasting on and off. No change.

Used a different slicer (SuperSlicer) for testing and the issue does not occur.

### Actual results

The prints have gaps between the walls.

### Expected results

No gaps between the walls.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[cura_for_mac_491_issue.zip](https://github.com/Ultimaker/Cura/files/6624440/cura_for_mac_491_issue.zip)

fvrmr, 2021-06-10:
Hi @renestrauch thank you for your report.
Did you try to change `fill gaps between walls` to everywhere?
If that doesn't help can you share a project file? (file - save project)
Your 3mf doesn't contain your settings.

renestrauch, 2021-06-10:
Yes, I have set the option Fill gaps between walls to everywhere.
[cura_project.3mf.zip](https://github.com/Ultimaker/Cura/files/6632033/cura_project.3mf.zip)

I create the 3mf again.

fvrmr, 2021-06-15:
Hi @renestrauch I can't see anything from the project file.
Looking at your photo it could be under extruding. Read here more about under extrusion: https://support.ultimaker.com/hc/en-us/articles/360012112859-How-to-fix-under-extrusion

Let me know if that helps for you.

renestrauch, 2021-06-15:
Thank you for the review
There is no under extrusion.
I tested everything also with settings for an over extrusion, which was also clearly seen. The gaps between the walls were still there.
As already written with another slicer I do not have the issue.
If there is no error in Cura, then I change the slicer.
This is not a problem for me.
Thanks for your time.

Ghostkeeper, 2021-06-17:
Your settings in Cura are different from whatever other slicer you used. After all, there is no 1:1 mapping between settings, and even for settings that do map, Cura's defaults will probably be different.

We can't help with underextrusion, since we don't have your printer to test with. We can't reproduce the issue then. We can only help with Cura's output being wrong.

From your project file, Cura's output looks fine with me. It's instructing the printer to extrude the correct amount of material in the correct spots. The fact that your printer doesn't properly execute those instructions then, is not something we can fix.

I suggest you read up on underextrusion and what to do about it. The Settings Guide also has a troubleshooting article on what you can do with Cura's settings to remedy underextrusion: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/underextrusion.md . I suggest you try lowering the print speed first.

I'll close this issue, since it's a problem with your print process, not a problem in Cura. If you have a suggestion on what we should change in the g-code output to remedy your problem, I'd still love to hear it of course!

closed 2021-06-17 by Ghostkeeper