Ultimaker/Cura #9956: cura 4.9.1 crash at startup

Manikem, 2021-06-09:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

None : first run

### Reproduction steps

Delete all previous cura instance
Delete all existing folders in program files and %appdata%
Install cura 4.9.1

### Actual results

After the launching popup indicating Launching UI a blank window appears for 2 seconds and disappears

### Expected results

App launch

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

log file
[cura.log](https://github.com/Ultimaker/Cura/files/6624542/cura.log)

fvrmr, 2021-06-10:
Hi @Manikem thank you for your bug report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9847 
Because it is a duplicate I will close this issue. For the next Cura release this will be fixed.

Manikem, 2021-06-10:
@fvrmr : I don't think this is caused by authentication there is no "cura.OAuth2.AuthorizationHelpers" in my cura.log

fvrmr, 2021-06-11:
You are right. But I can't see anything else that would cause this. 
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-06-15:
Hi @Manikem we can't reproduce your issue. Do you have maybe an external monitor?

Manikem, 2021-06-15:
Hello yes I have an external monitor but cura appears on my main screen at startup. 
How can I help you ?
Reinstall last version of cura, try to launch with my external monitor disconnected ?

fvrmr, 2021-06-15:
Yes, try to launch Cura without your external monitor.

Ghostkeeper, 2021-06-15:
If the crash happens due to the extra monitor, we might have a fix for that in Cura 4.10.

Manikem, 2021-06-16:
Hello due to multiple test 1 don't know exactly what was the lastest version I was able to run so I download all version from 4.6.1 to 4.9.1 ans with my external screen unplugged.
The best version which run is 4.7.1,  the greater one with "win64" in filename.
All version with amd64 in the filname doesn't work.

[image]

Ghostkeeper, 2021-06-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/7183 then.

closed 2021-06-17 by Ghostkeeper

Manikem, 2021-06-17:
Reinstalled 4.9.1 to try to correct cura.cfg as specified in #7183, but always crashing at startup, the config file allready has the option set to false. joined my cfg file (not edited).
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/6669128/cura.cfg.txt)

fvrmr, 2021-06-21:
Hi @Manikem Cura 4.10 beta is released last Friday. Could you try to this version and see if Cura doesn't crash anymore on start up.
https://github.com/Ultimaker/Cura/releases/tag/4.10-beta