Ultimaker/Cura #9958: Program crashes after selecting the zoom menu

tyrpyr-true, 2021-06-09:
### Application Version

4.9.1

### Platform

Linux 5.4.0-74-generic #83-Ubuntu  x86_64 x86_64 x86_64 GNU/Linux

### Printer

Anycubic Mega X

### Reproduction steps

1. open the stl project file
2. select the model on the table
3. click on the zoom menu

### Actual results

the program closes, the error is not displayed. if you run it from the terminal, the following is output:
user001@user001-comp:~/Cura/bin$ ./4.9_20210511_Ultimaker_Cura-4.9.1.AppImage
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Cyclic dependency detected between "file:///tmp/.mount_4.9_20WC9Gpf/usr/bin/resources/qml/Actions.qml" and "file:///tmp/.mount_4.9_20WC9Gpf/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/tmp/.mount_4.9_20WC9Gpf/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
QOpenGLShader::compile(Fragment): 0:14(34): error: syntax error, unexpected ';', expecting '{'

*** Problematic Fragment shader source code ***
#ifdef GL_KHR_blend_equation_advanced
#extension GL_ARB_fragment_coord_conventions : enable
#extension GL_KHR_blend_equation_advanced : enable
#endif
#define lowp
#define mediump
#define highp
#line 1

uniform mediump vec4 u_ambientColor;
uniform mediump vec4 u_diffuseColor1;
uniform mediump vec4 u_diffuseColor2;
uniform mediump vec4 u_specularColor;
uniform mediump float u_opacity;
uniform highp vec3 u_lightPosition;
uniform mediump float u_shininess;
uniform highp vec3 u_viewPosition;

uniform mediump float u_width;
uniform bool u_vertical_stripes;
uniform lowp u_lowestPrintableHeight;

varying highp vec3 v_position;
varying highp vec3 v_vertex;
varying highp vec3 v_normal;

void main()
{
mediump vec4 finalColor = vec4(0.0);
mediump vec4 diffuseColor = u_vertical_stripes ?
(((mod(v_vertex.x, u_width) < (u_width / 2.)) ^^ (mod(v_vertex.z, u_width) < (u_width / 2.))) ? u_diffuseColor1 : u_diffuseColor2) :
((mod(((-v_vertex.x + v_vertex.y + v_vertex.z) * 4.), u_width) < (u_width / 2.)) ? u_diffuseColor1 : u_diffuseColor2);

/* Ambient Component */
finalColor += u_ambientColor;

highp vec3 normal = normalize(v_normal);
highp vec3 lightDir = normalize(u_lightPosition - v_vertex);

/* Diffuse Component */
highp float NdotL = clamp(abs(dot(normal, lightDir)), 0.0, 1.0);
finalColor += (NdotL * diffuseColor);

/* Specular Component */
/* TODO: We should not do specularity for fragments facing away from the light.*/
highp vec3 reflectedLight = reflect(-lightDir, normal);
highp vec3 viewVector = normalize(u_viewPosition - v_vertex);
highp float NdotR = clamp(dot(viewVector, reflectedLight), 0.0, 1.0);
finalColor += pow(NdotR, u_shininess) * u_specularColor;
if (v_vertex.y <= u_lowestPrintableHeight)
{
finalColor.rgb = vec3(1.0, 1.0, 1.0) - finalColor.rgb;
}

gl_FragColor = finalColor;
gl_FragColor.a = u_opacity;
}
***
QOpenGLShader::compile(Fragment): 0:14(34): error: syntax error, unexpected ';', expecting '{'

*** Problematic Fragment shader source code ***
#ifdef GL_KHR_blend_equation_advanced
#extension GL_ARB_fragment_coord_conventions : enable
#extension GL_KHR_blend_equation_advanced : enable
#endif
#define lowp
#define mediump
#define highp
#line 1

uniform mediump vec4 u_ambientColor;
uniform mediump vec4 u_diffuseColor1;
uniform mediump vec4 u_diffuseColor2;
uniform mediump vec4 u_specularColor;
uniform mediump float u_opacity;
uniform highp vec3 u_lightPosition;
uniform mediump float u_shininess;
uniform highp vec3 u_viewPosition;

uniform mediump float u_width;
uniform bool u_vertical_stripes;
uniform lowp u_lowestPrintableHeight;

varying highp vec3 v_position;
varying highp vec3 v_vertex;
varying highp vec3 v_normal;

void main()
{
mediump vec4 finalColor = vec4(0.0);
mediump vec4 diffuseColor = u_vertical_stripes ?
(((mod(v_vertex.x, u_width) < (u_width / 2.)) ^^ (mod(v_vertex.z, u_width) < (u_width / 2.))) ? u_diffuseColor1 : u_diffuseColor2) :
((mod(((-v_vertex.x + v_vertex.y + v_vertex.z) * 4.), u_width) < (u_width / 2.)) ? u_diffuseColor1 : u_diffuseColor2);

/* Ambient Component */
finalColor += u_ambientColor;

highp vec3 normal = normalize(v_normal);
highp vec3 lightDir = normalize(u_lightPosition - v_vertex);

/* Diffuse Component */
highp float NdotL = clamp(abs(dot(normal, lightDir)), 0.0, 1.0);
finalColor += (NdotL * diffuseColor);

/* Specular Component */
/* TODO: We should not do specularity for fragments facing away from the light.*/
highp vec3 reflectedLight = reflect(-lightDir, normal);
highp vec3 viewVector = normalize(u_viewPosition - v_vertex);
highp float NdotR = clamp(dot(viewVector, reflectedLight), 0.0, 1.0);
finalColor += pow(NdotR, u_shininess) * u_specularColor;
if (v_vertex.y <= u_lowestPrintableHeight)
{
finalColor.rgb = vec3(1.0, 1.0, 1.0) - finalColor.rgb;
}

gl_FragColor = finalColor;
gl_FragColor.a = u_opacity;
}
***
Only C and default locale supported with the posix collation implementation
Only C and default locale supported with the posix collation implementation
Case insensitive sorting unsupported in the posix collation implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation implementation
Only C and default locale supported with the posix collation implementation
Case insensitive sorting unsupported in the posix collation implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation implementation
Only C and default locale supported with the posix collation implementation
Case insensitive sorting unsupported in the posix collation implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation implementation
Only C and default locale supported with the posix collation implementation
Case insensitive sorting unsupported in the posix collation implementation
Numeric mode unsupported in the posix collation implementation
Illegal instruction (core dumped)
user001@user001-comp:~/Cura/bin$

### Expected results

the zoom menu opens

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6625391/cura.log)

[project.zip](https://github.com/Ultimaker/Cura/files/6625413/project.zip)

df9ry, 2021-06-09:
Confirmed. Same misbehavior, exactly the same message and coredump. 
Linux Latitude-E6420 5.4.0-74-generic #83-Ubuntu SMP Sat May 8 02:35:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

smartavionics, 2021-06-09:
Duplicate of #9692. This has been fixed for a future UM release. In the meantime, you could use one of my builds which already have this issue fixed. You can find my builds at https://github.com/smartavionics/Cura/releases.

fvrmr, 2021-06-10:
Closing this issue because it is a duplicate of #9692

closed 2021-06-10 by fvrmr