Ultimaker/Cura #9959: Small Feature Speed option does not work

asychev, 2021-06-09:
### Application Version

4.9.1

### Platform

macOS Big Sur 11.4

### Printer

Creality Ender 3 V2

### Reproduction steps

Print Speed - 50 mm/s
Initial Layer Speed - 10 mm/s
Small Hole Max Size - 3.0 mm
Small Feature Speed - 10%
Small Feature Initial Layer Speed - 10%

### Actual results

Print speed for whole initial layer set to 10 mm/s (= Initial Layer Speed)

### Expected results

Small parts on initial layer should print with 50 * 0.1 (10%) = 5 mm/s

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

asychev, 2021-06-09:
Maybe I understand this option in a wrong way? Should lizard fingers print with slower speed?

smartavionics, 2021-06-09:
What's the min-speed value in the cooling section? It won't use a speed less than that.

asychev, 2021-06-09:
@smartavionics Thanks, it was set to 10 mm/s indeed (probably, equal to Initial Layer Speed by default?). Unfortunately, change does not give required results:

[image]

What I'm trying to achieve - improvement of small details adhesion on initial layer. Long lines print well, but small holes or details (like fingers here) usually does not stick and create blobs. Seems like this is not possible?

smartavionics, 2021-06-09:
But the fingers are not small features. Sure, they aren't very wide but the total length of the wall that goes around the fingers is much longer than the small feature line length. Anyway, if you are still getting poor adhesion at 10mm/S, there's something else wrong. I never need to go slower than that when printing with PLA/PETG/TPU no matter how small an object I am printing. The obvious things to check are the build plate surface, the first layer height, suitable nozzle and bed temperatures, extruder calibration...

asychev, 2021-06-09:
@smartavionics Thanks for the clarification.

closed 2021-06-09 by asychev