Ultimaker/Cura #9960: Tool change "optimization" enable/disabbutton

colin-donar, 2021-06-09:
### Is your feature request related to a problem?

I have multiple extruders but I always want T1 to start the layer. In the background CURA optimizes tool changes by alternating what tool starts first, thus cutting tool changes in half. This not what I want. It would be preferred if I could  specify what tool starts the layer, even better if I could control the order of multiple (>2) tool. 

What I need:
Layer 1:
T1 T0
Layer 2: 
T1 T0
Layer 3:
T1 T0

What CURA does:
Layer1:
T0 T1
Layer 2:
T1 T0 
Layer 2:
T0 T1

### Describe the solution you'd like

It would be preferred if I could  specify what tool starts the layer, even better if I could control the order of multiple (>2) tools. 

### Describe alternatives you've considered

A simple check box to disable the tool changing optimization

### Affected users and/or printers

Custom FFF  & E3D toolchanger

fvrmr, 2021-06-11:
Hi @colin-donar thank you for your feature request.
Could you tell me in what case you want to use this?

colin-donar, 2021-06-11:
My use case is rather specific but I think that this would be applicable to any multi tool machines.

Essentially what I have is a multi-material print and I want each layer to start with the same material. In my case whatever material gets laid down first spills, for lack of a better word, into where the second material would be. And when Cura alternates what material starts each layer I get a rougher surface quality, between the two materials, that I would have if the first material would have started each layer.

[image]
In layer 1 blue material was deposited first then orange
Layer 2 orange then blue
etc.
I don't have an  IRL image but I have seen this result in person. Especially noticeable with large layer heights

Another more generalized example could be if I'm concentrically printing one material inside another. Theoretically it would be best to print the inside material first each layer and then the outside shell so that the inside material nozzle doesn't need to travel over the outside material. Z-hop only helps to an extent...

I would like an option to tell Cura to start each layer with the same extruder. Even better if this feature could handle ordering more than two extruders.

fvrmr, 2021-06-15:
Thank you, I will bring this up with the team. Keep you posted.

Ghostkeeper, 2021-06-16:
I know of the effect. The material that gets deposited first has more space to flow, so it will flow out more.

However I disagree with the use case, because of this sentence:

> And when Cura alternates what material starts each layer I get a rougher surface quality, between the two materials, that I would have if the first material would have started each layer.

Between the two materials, you want to get a somewhat rough surface quality, so that the materials attach better together. Why would you want the surface between two materials to be smooth on the inside of the model?

colin-donar, 2021-06-16:
I mentioned my use case is rather specific... I am actually trying to deposit powder and the second material would be removed after curing, exposing the seam between the materials.  
However I think that this feature would also be useful for complex dissolvable supports.

Also I think my second case still stands
> Another more generalized example could be if I'm concentrically printing one material inside another. Theoretically it would be best to print the inside material first each layer and then the [...]

Ghostkeeper, 2021-06-16:
For the extruder switch most printers would have to move over the concentric outer shell anyway, since they either need to move the new nozzle in position (a move by the nozzle offset at best, or all the way to the switching bay at worst), or they'd need to purge in a prime tower in case of a shared-nozzle design.

For soluble supports it is better to not let the support touch the side of the model, since that greatly increases the time it takes for the material to dissolve, and prevents the possibility of combing which also reduces print quality.

We've decided to reject this feature, since the hazards of misuse are considered worse than the benefits of this small use case. Sorry!

closed 2021-06-16 by Ghostkeeper