Ultimaker/Cura #9961: Support Blocker Modify settings for overlaps (cutting mesh) makes wall width thin

Screlias, 2021-06-10:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Insert support blocker
2. Size it and move it where I wanted it
3. Per Model Settings - Modify settings for overlaps - cutting mesh and nothing else (I wanted to use fuzzy skin)
4. Slice

### Actual results

[image]
[image]
[image]
and I tried it on another model:
[image]

### Expected results

The walls should have stayed at the same width as well the bottom layers 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Same steps and profile and blocker placement in cura 4.4.1 no problems there...
[image]
[image]
There is definitely a bug or I am doing something very wrong

fvrmr, 2021-06-10:
Hi @Screlias thank you for your report.
Could you share your project file (file - save project) than I can take a look at it.

Screlias, 2021-06-11:
Without changing anything (except downloading cura 4.4.1) it doesn't do it any more! And I spent hours searching for a solution and all it needed was a computer restart -_- ! Although it bugged for sure, I didn't notice it until the print broke on the 67th layer where the infill line was so thin the didn't stick together. I am sorry I can't reproduce the bug because it is a major issue, which turned the print to rubbish. 
[image]
[image]
The project file has the same settings...
[CE3_fem_pic_rail_diam_gripv2.zip](https://github.com/Ultimaker/Cura/files/6636197/CE3_fem_pic_rail_diam_gripv2.zip)

fvrmr, 2021-06-11:
Thank you for your project file. I think this was caused by enabling the bridging setting. Because when I enabled it shows exactly the same as you screenshots.

[image]

But you did make it work now in 4.9.1 right?

Screlias, 2021-06-11:
Yes you are right I forgot that I had enabled bridging but is it normal?

fvrmr, 2021-06-15:
I would say no, because there is no need for bridging. It could be that fuzzy skin and bridging doesn't work nice together. 
I will bring it up with the team. Keep you posted!

Ghostkeeper, 2021-06-16:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9083. Please continue the conversation there.

closed 2021-06-16 by Ghostkeeper

Screlias, 2021-06-16:
It has nothing to do with fuzzy skin. The bug occurs when you have bridging enabled and you insert a support blocker in cutting mesh mode.

Ghostkeeper, 2021-06-17:
Yeah I think the "fuzzy skin" was just a mix-up. fvrmr meant to say Cutting Mesh. And indeed, they don't work nice together. We consider that to be a bug.