Ultimaker/Cura #9962: PyCapsule_GetPointer called with incorrect name

frank-pian, 2021-06-10:
### Application Version

master

### Platform

ubuntu 20.04

### Printer

none

### Reproduction steps

1. `sudo apt-get install python3-scipy python3-pip
sudo pip3 install pyserial PyQt5==5.15 shapely zeroconf trimesh`
2. Build and install SIP 4.19.25
3. Build and install libArcus 4.9.1
4. Build and install pynest2d master
5. run cura_app.py

### Actual results

ValueError: PyCapsule_GetPointer called with incorrect name

If I'm using PyQt 5.10, report error "No module PyQt5.sip"

### Expected results

Run Cura normally

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

`/home/frankpian/develop/Uranium/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses

 import imp

Error in sys.excepthook:

Traceback (most recent call last):

 File "<frozen importlib._bootstrap>", line 991, in _find_and_load

 File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked

 File "<frozen importlib._bootstrap>", line 671, in _load_unlocked

 File "<frozen importlib._bootstrap_external>", line 783, in exec_module

 File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed

 File "/home/frankpian/develop/Cura/cura/CuraApplication.py", line 35, in <module>

   from UM.Qt.QtApplication import QtApplication # The class we're inheriting from.

 File "/home/frankpian/develop/Uranium/UM/Qt/QtApplication.py", line 19, in <module>

   from UM.Backend.Backend import Backend #For typing.

 File "/home/frankpian/develop/Uranium/UM/Backend/Backend.py", line 12, in <module>

   from UM.Backend.SignalSocket import SignalSocket

 File "/home/frankpian/develop/Uranium/UM/Backend/SignalSocket.py", line 4, in <module>

   import Arcus

ValueError: PyCapsule_GetPointer called with incorrect name

Original exception was:

Traceback (most recent call last):

 File "cura_app.py", line 187, in <module>

   from cura.CuraApplication import CuraApplication

 File "/home/frankpian/develop/Cura/cura/CuraApplication.py", line 35, in <module>

   from UM.Qt.QtApplication import QtApplication # The class we're inheriting from.

 File "/home/frankpian/develop/Uranium/UM/Qt/QtApplication.py", line 19, in <module>

   from UM.Backend.Backend import Backend #For typing.

 File "/home/frankpian/develop/Uranium/UM/Backend/Backend.py", line 12, in <module>

   from UM.Backend.SignalSocket import SignalSocket

 File "/home/frankpian/develop/Uranium/UM/Backend/SignalSocket.py", line 4, in <module>

   import Arcus

ValueError: PyCapsule_GetPointer called with incorrect name`

nallath, 2021-06-10:
Did you try to build Arcus on master? Your report mentions that you used 4.9.1. It should work if all branches are on master (since we recently fixed the exact issue that you are having)

frank-pian, 2021-06-15:
> 
> 
> Did you try to build Arcus on master? Your report mentions that you used 4.9.1. It should work if all branches are on master (since we recently fixed the exact issue that you are having)

it 's work on master branch.

closed 2021-06-15 by frank-pian