Ultimaker/Cura #9963: Segmentation fault

moose4621, 2021-06-10:
### Application Version

4.9_20210511

### Platform

Ubuntu 20.04.2 LTS 64-bit

### Printer

Prusa i3

### Reproduction steps

After some incremental updates to Ubuntu this morning, Cura app image now stops after loading a stl model and then trying to use the edit menu, or right clicking on model.
I am running the 4.9_20210511_Ultimaker_Cura-4.9.1.AppImage by double clicking on the file. I also tried running it direct from command line and get the following errors.
`Inspiron-5593:~$ Cura/4.9_20210511_Ultimaker_Cura-4.9.1.AppImage 
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Cyclic dependency detected between "file:///tmp/.mount_4.9_20aQLNFK/usr/bin/resources/qml/Actions.qml" and "file:///tmp/.mount_4.9_20aQLNFK/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/tmp/.mount_4.9_20aQLNFK/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
Segmentation fault (core dumped)
Inspiron-5593:~$`
Not sure how to list the updates this morning to the OS and software.

### Actual results

Cura app image now stops after loading a stl model and then trying to use the edit menu, or right clicking on model.

### Expected results

Cure was working as expected yesterday and previous to then. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura_log.txt](https://github.com/Ultimaker/Cura/files/6630594/cura_log.txt)

fvrmr, 2021-06-11:
Hi @moose4621 thank you for your bug report.
I will bring it up with the team. Keep you posted!

moose4621, 2021-06-13:
After installing new updates for Ubuntu, the problem is gone again. Seems it was a bug in the OS. My world is at peace again. :-)

closed 2021-06-13 by moose4621

yash-fn, 2021-07-21:
I am still having this problem. I updated system software but it is still not working. Any ideas?

Ghostkeeper, 2021-07-28:
> I am still having this problem. I updated system software but it is still not working. Any ideas?

Even if you have the same symptom, if the solution to an issue doesn't work for you, you have a different issue. If you post a new issue on Github with proper reproduction steps and a log, we may be able to help out.