Ultimaker/Cura #9965: Cura 4.9.1 display bug.

kangoam, 2021-06-10:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

hello,
1. I open the print settings and click on the small wheel for the "Fill" settings.

[image]

### Actual results

1. The settings window is systematically placed at the top of the list, displaying the "Quality" section.

[image]

### Expected results

1. Cura should display the "Filling" section directly.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Can the problem be caused by a third-party Windows software?

Thanks for your help, and sorry for my very poor English!

fvrmr, 2021-06-11:
Hi @kangoam thank you for your bug report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9621 and this is fixed for the next release. 
I will close this because it is a duplicate.

closed 2021-06-11 by fvrmr

kangoam, 2021-06-11:
@fvrmr  Thank you so much for giving me the link to the post solving my problem!!!