Ultimaker/Cura #9966: Cura 4.9.1 crashes on startup after swichting GPU

chiefmaster81, 2021-06-11:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

After switching my GPU to NVIDIA 3070 ( from 3060 ) Ultimaker crashes on startup.
I tried Windows 10 compatibility modes and fresh install of NVIDIA drivers an CURA.

Problem persists.

### Actual results

Cura crashes

### Expected results

cura should start normally.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/6636677/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/6636679/cura.log)

fieldOfView, 2021-06-11:
> Cura crashes

Please be more specific. Do you get an error message? Does the Cura window just disappear? Does the window "lock up" in any way? If you are more specific, chances of use being able to fix your issue increase.

chiefmaster81, 2021-06-11:
ok, sorry. Cura Loading scrren appears, loads some modules and then simply disappears.

fvrmr, 2021-06-11:
Hi @chiefmaster81 thank you for your report. I see that you also commented on https://github.com/Ultimaker/Cura/issues/9940
And this issue is a duplicate of that one, I will close this is as a duplicate. So we keep everything in at one place.

closed 2021-06-11 by fvrmr