Ultimaker/Cura #9967: Marketplace causes hang on MacOS

jswadley, 2021-06-11:
### Application Version

4.9.1

### Platform

MacOS 11.4

### Printer

Voxelab Aquila

### Reproduction steps

Open the Marketplace.

### Actual results

The entire app hangs for 10-15 seconds and makes the whole app unresponsive.

### Expected results

Marketplace opens instantly like any other part of the app.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Thanks!

fvrmr, 2021-06-11:
Hi @jswadley thank you for your report.
Could you also share your log file? You can find your log file here: `$USER/Library/Application Support/cura/<Cura version>/cura.log`

jswadley, 2021-06-11:
Sure! Sorry, wasn't sure it'd be useful.
[cura.log](https://github.com/Ultimaker/Cura/files/6638735/cura.log)

fieldOfView, 2021-06-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/7071. It is not just MacOS.

nallath, 2021-06-11:
We're planning to do a full refactor of the marketplace soon. The slowness is one of the main reasons for it. So no worries, we expect to fix this soon!

closed 2021-06-11 by nallath