Ultimaker/Cura #9968: improve buildtimes

anonhelper1337, 2021-06-11:
### Is your feature request related to a problem?

I'd like to bring an idea to introduce a new feature to improve print speed and potentiall improve the stabilty of the print.

I hate to have a bad resolution on my prints(layer height 0.2) and I love the quality of a 0.04 layer height 3d print. However it's sad that a print of something mid-sized just takes a week to print or even longer. After all, you see only the outside and the layerheight should be changeable between the outside and the inside(infill !!!) of the print

### Describe the solution you'd like

introducing a feature/setting, that does only print outer walls with 1 or 2 layers in the resolution you choose(example 0.04) and give the possiblity to increase the height of the rest of the perimiters to something higher? same thing for the infill. the infill benefits under certain circumstances from overextrusion(best would be to change the line width)/higher layers

The result should be for example;
I have for example 4 wall line counts and some infill;
 - outer walls(1 or 2 wall lines, should be configurable in the best case) with height 0.04 , line width 0.4 printed
 - the remaining walls can be printed with a layer hight of 0.2
 - the infill can also be printed with a worse resolution heigh, 0.4 and line witdh of 0.6(works with a 0.4 nozzle ok. if you're pushing it, you can go up to 0.8 but you will need to play with the extrusion temperature...) the line width can possibly be also achieved by upping the "infill flow" but the layer height obiosuly not...

With the above the build speed would benefit and the build quality should IMHO benefit as well as you can use a much besser resolution on the parts, that you actually see..

### Describe alternatives you've considered

none, I don't see a practical way of having different layer heights in your build.
the only thing I can think of, is generating the gcode by cura, write a python script, that does then reprocess the gcode with the above strategy(deleting every x amount of line of the infill, upping the flow ratio while working on the infill)..
 
However I'd hate to create such a soution just for myself. Since I believe if this whole thing works out it would really help to improve 3d printing.****

### Affected users and/or printers

hopefully everybody and will revolutionize the printing speeds for everybody in 3d printing community :)

fieldOfView, 2021-06-11:
This already exists for infill (see "Infill Layer Thickness") and support (see " "Support Infill Layer Thickness"), but not yet for inner walls.

Ghostkeeper, 2021-06-21:
The remainder is a duplicate of this feature request: https://github.com/Ultimaker/Cura/issues/6322

closed 2021-06-21 by Ghostkeeper