Ultimaker/Cura #9969: Travels from Support to Support Don't Obey Combing modes.

3Duke3, 2021-06-11:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

All models (specifically Creality CR10 S5)

### Reproduction steps

Set combing mode to :Within infill (or All, or Not in skin )

### Actual results

Support to support travel moves are performed through open areas of part geometry (ie nozzle travels through multiple print walls) despite obvious travel paths contained within the infill of part. Appears to only happen when traveling from support to support, any part related travel move obeys the "within infill" rule.

After printing an area of support, the nozzle makes a direct shortest route travel move to the next area of support, crossing whatever part geometry is in its path.

Understandably in this case the nozzle must cross walls to reach the desired area of support, however once it crosses one wall, it should then obey travel within infill, untill it need to cross the next wall to reach more support.

Black indicates the current support to support travel path, Purple would be a more sensible route suggestion.

[image]

### Expected results

After printing an area of support, the nozzle should travel to the next area of support, crossing the least amount of print walls possible, while attempting to travel the shortest path.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Support to Support Travels dont obey Combing modes.zip](https://github.com/Ultimaker/Cura/files/6641016/Support.to.Support.Travels.dont.obey.Combing.modes.zip)

fvrmr, 2021-06-14:
Hi @3Duke3 thank you for your report.
I will discuss this with the team. Keep you posted!

fvrmr, 2021-06-15:
I have discussed this with the team and I have created a ticket on our backlog for this issue.
Devs see CURA-8319

nallath, 2021-11-05:
I was looking through our backlog and came across this issue. I think the behavior is actually correct here.

The infill is printed first, so at that moment there is no other stuff on that layer to comb through. As such it will just take the fastest path.

mattwn, 2022-01-20:
I have also noticed this behavior with combing as it transitions from ;TYPE:SUPPORT to ;TYPE:OUTER-WALL or similar. I believe it may related to OP's issue.

Regarding the comment "The infill is printed first, so at that moment there is no other stuff on that layer to comb through. As such it will just take the fastest path." This might not always be the case. In the screenshot below (from the project also included below), you can see the intra-support travel move (top right of screenshot) happens before the outer wall is printed, yet still plans a route that avoids where the wall will be rather than take the fastest direct path over the open space.

This is Layer 17 in Cura Preview (so something around layer 11 in G Code):

[image]

Every other transition of G Code section has the proper combing behavior. But the ;MESH:FILENAME section, which serves as the transition after a ;TYPE:SUPPORT section seems potentially bugged.

This screenshot is from a very simple project to demonstrate the potential bug and is included below. 

[CE3_VikinigHorn.zip](https://github.com/Ultimaker/Cura/files/7908833/CE3_VikinigHorn.zip)

mattwn, 2022-01-21:
If this is indeed intended behavior (because the "wall-outer" has not been printed yet), then consider a feature request to all calculate future polygons when combing.

Looking briefly through the CuraEngine code, I mightassume the wall-outer is not in the list of polygons that LinesPolygonsCrossing is looking at, because it has not been printed yet? Although that seems incorrect because the intra-support travel move (top right corner of the graphic in previous reply) clearly took a path to avoid the not-yet-printed Wall-Outer. So that implies LinesPolygonsCrossing already had knowledge of a not-yet-printed polygon on the same layer.

closed 2024-12-09 by github-actions[bot]