Ultimaker/Cura #9970: Cura not Booting

markbot2000, 2021-06-12:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Opened Cura

### Actual results

Cura freezes on "Initializing build volume" on boot

### Expected results

Cura should load

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.zip](https://github.com/Ultimaker/Cura/files/6641534/cura.zip)
Log File

Screenshots
[image]
[image]

markbot2000, 2021-06-12:
[Traceback.zip](https://github.com/Ultimaker/Cura/files/6641589/Traceback.zip)
Traceback file

fieldOfView, 2021-06-12:
This crash seems to be caused by you having logged in to an Ultimaker account before, and now Cura not being able to contact the ultimaker server to confirm your account status. That bug has been fixed for the next release of Cura (https://github.com/Ultimaker/Cura/pull/9903).

To solve it for you, either make sure the computer that runs Cura is connected to the internet, or:
* open `%APPDATA%\cura\4.9\cura.cfg` in a texteditor (notepad will do)
* remove the line that starts with `ultimaker_auth_data =`
* save the file and restart Cura

This will log you out of your Ultimaker account, but that will not affect the functionality of Cura much.

fieldOfView, 2021-06-12:
Duplicate of https://github.com/Ultimaker/Cura/issues/9847

fvrmr, 2021-06-14:
Closing this issue since it is a duplicate of #9847

closed 2021-06-14 by fvrmr