Ultimaker/Cura #9971: Cura crashes

grimm1007, 2021-06-12:
### Application Version

4.9.1

### Platform

Windows 10 , gpu gtx750

### Printer

Ender 3, Creality

### Reproduction steps

1) clicked on folder button
2) entered a folder (1)
3)entered another folder within the folder (1 - 1)
4) entered folder within folder within folder (1 -1 - 1 )

### Actual results

cura crashes when I try to access a folder within 3 other folders

### Expected results

open folder to access 3d models

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Can't seem to find the crash log for cura

fvrmr, 2021-06-14:
Hi @grimm1007 thank you for your report.
Do you have Chitubox installed?

grimm1007, 2021-06-14:
Yes I do. Does it affect cura ?

closed 2021-06-14 by fvrmr