Ultimaker/Cura #9973: Support blockers get grouped upon loading the project, ungrouping them sets Z to 0.

Hologos, 2021-06-12:
### Application Version

4.9.1

### Platform

macOS Catalina 10.15.7

### Printer

Creality Ender 3 V2

### Reproduction steps

1. Load model
2. Add support blocker
3. Save the project
4. Close Cura
5. Open cure
6. Load the project
7. Ungroup the model and support blocker

### Actual results

Z position of all support blockers gets set to 0.

### Expected results

They should stay, where they were.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Here is [a video](https://user-images.githubusercontent.com/1620004/121787429-45cda400-cbc6-11eb-9dd0-8f684c37ff3d.mov).
Here is [the cura.log](https://github.com/Ultimaker/Cura/files/6642802/cura.log).

GregValiant, 2021-06-13:
It appears to be OK in Windows 10.  To select a single item from a group in Windows it's CNTRL+LeftClick.  That works too.

nallath, 2021-06-14:
I've tried this on Linux, it also doesn't load it back as a group. Perhaps it's a OSX only issue?

fvrmr, 2021-06-14:
I've tried it on macOS Catalina 10.15.7 and couldn't reproduce this. Can you share your project file as a zip?

Hologos, 2021-06-16:
Hi, I am sorry for not including the files. I had them ready and forgot to include them.
[2021-06-12.bug-report.support-blocker-z-pos-resets.zip](https://github.com/Ultimaker/Cura/files/6665580/2021-06-12.bug-report.support-blocker-z-pos-resets.zip)

fvrmr, 2021-06-17:
Thank you I can now reproduce this on Windows. I will discuss this with the team. Keep you posted!

Hologos, 2021-06-17:
Here is a list of 3rd party plugins I have installed.
[image]

closed 2021-11-05 by nallath