Ultimaker/Cura #9975: Add a "preference" to add material name prefix to job name

stevenfayers, 2021-06-14:
### Is your feature request related to a problem?

The Gcode file name generated contains the machine prefix automatically, but not the material prefix. It would be great to have the material prefix (such as PLA or ABS) to known which material I had selected when slicing. This is more useful that the machine name when most people have one machine.

### Describe the solution you'd like

Ability to prefix job name with material name such that job123 would have a filename ABSjob123.gcode or PLAjob123.gcone

### Describe alternatives you've considered

Manually change the filename everytime

### Affected users and/or printers

All users that print with multiple materials

Asterchades, 2021-06-14:
This functionality (and more) are already present in the "GCode Filename Format" plugins.

fieldOfView, 2021-06-14:
... and the Printjob Naming plugin.

fvrmr, 2021-06-14:
Closing this feature request, because this can be done with plugins that are mentioned above.

closed 2021-06-14 by fvrmr