Ultimaker/Cura #9976: Infill not printing at specified width

quadcom, 2021-06-14:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Anycubic Mega Zero v2

### Reproduction steps

I've been seeing infill that is not continuously joined as if extrusion is too low. But I noticed that support material didn't have this issue. 

I measured the support material which was 0.4mm. But infill was closer to 0.2mm even though both support and infill were set to 0.4mm

Printing the owl.gcode included with the printer does not show this behaviour so I don't see this as a printer-specific issue. I am still new to printing and I have a lot to learn but this seemed fishy.

[image]

[image]

### Actual results

Infill is breaking up

### Expected results

Solid grid infill without breakups.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Among Us 400pct.zip](https://github.com/Ultimaker/Cura/files/6648838/Among.Us.400pct.zip)

fvrmr, 2021-06-15:
Hi @quadcom thank you for your report. 
You could prevent this by set the infill speed slower or change the pattern to lines.
I have to defer this bug because we don't have anytime soon to be picked up and it would also be a bit complex to solve. 

Feel free to reach out if it isn't clear to you.

quadcom, 2021-06-15:
@fvrmr - Thanks for checking in. I need to point out that both support material and infill were set to the same width and printed at the same speed. I initially had infill set to lines and it was even worse. Worse because the gap under the extruder is larger when using lines which means the line width will actually be smaller. There's nothing to 'squish' the material to create the width.

I filed this as a potential bug because both support and infill were printed with exactly the same settings but the resulting output was dramatically different. As I mentioned, I did measure, with callipers, the line width of both support and infill. Even the infill in areas that were solid and not broken up, the width was not what was set.

Hopefully this gets picked up and checked but I understand priorities.

Ghostkeeper, 2021-06-22:
When I saw this issue I thought it was just a settings problem, underextrusion caused by infill lines crossing. The problem is that the backpressure caused by the crossing of an infill line stops the flow of plastic, and this takes a moment to restart again. There are different infill patterns that don't cross: lines, gyroid, or (ironically) cross infill.

After discussing why it was deferred if it was just a settings problem, it turns out this ticket was adjusted to a feature request to deal with crossing infill lines better. I'll adjust the title and tags to match that.

quadcom, 2021-06-22:
@Ghostkeeper Originally, I was using lines as the infill but that was even worse. I suspected the lines were so thin was due to there being nothing underneath the nozzle. The infill was one big constant bridge. That caught me off guard as the first print I ever made, the owl, had a grid infill. But, that infill didn't have the issues I see here. That's one of the reasons why I posted this ticket.

I'm case it's missed, the owl gcode was included in the printer (Anycubic). It was not (afaik) sliced in Cura. At least not by me.  Slices I make in Cura seen to all have this kind of issue.

Ghostkeeper, 2021-06-23:
The lines pattern alternates by default, but doesn't have to do that. If you specify the angles of the lines in the Infill Line Directions setting you can make it stop alternating, or alternate in different directions.

The issue reported here is that the line gets a gap just after every crossing. If the entire infill is one big constant bridge, at least the infill lines will be consistently thin, so that would be a different issue.

Note that generic underextrusion is not something we can really help with. It's specific to your printer and should be solved in that printer's profiles. Also, the infill line width being 100% accurate is not a goal in itself. The material usage should be accurate, and the print should be stiff or otherwise strong, but material will flow around based on all sorts of factors which are not necessarily always bad. If it's "always bridging" that effectively means the line gets twice as much vertical space, similar to Infill Layer Thickness, which means that it'll end up being more round than flat which can be a good property too.