Ultimaker/Cura #9979: Cura adding bridges that aren't in the STL

musicCodeGuy, 2021-06-15:
### Application Version

4.9.1

### Platform

up to date arch linux x86-64

### Printer

creality ender 5 and sovol sv01

### Reproduction steps

Load the attached project
[CE5_ebikeBatteryCase3.zip](https://github.com/Ultimaker/Cura/files/6652158/CE5_ebikeBatteryCase3.zip)

. Look inside the box. Note it's empty. Slice it. 

### Actual results

Look inside in preview mode and you find cura sprayed garbage into the model. Problem seems to vanish at 0.1mm layer height but is there at 0.2. Different materials and different printers make no difference
[image]

### Expected results

The box should be empty as it is in the model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

There's no difference in results between the appimage and repository packages.

PrusaSlicer outputs perfect gcode from the same model and gave me a successful print.

fvrmr, 2021-06-15:
Hi @musicCodeGuy thank you for your report.
I can't reproduce your issue. Sliced your project file and there is no extra layer in the model.

[image]

musicCodeGuy, 2021-06-15:
@fvrmr so it ISN'T an intel floating point bug as it does exactly the same thing on the broadwell laptop. I won't be able to test on the rpi's until tomorrow my time as my main rpi is engaged in several large prints and I don't want to risk slicing while running 3 octoprint instances or bringing up my 2nd rpi with the same host name from a backup.

Given it happens on two different intel CPUs from very different eras and it happens with the appimage as well as the normal arch linux repo packages, it's almost certainly an issue between cura and the current arch linux glibc which I've just checked to be version 2.33-5, as appimages rely on the host system's glibc and that's the only difference assuming you tried slicing on a linux box.

It's an up to date manjaro-arm I have on the rpi running the multiple  octoprint instances, I'll try a slice on it tomorrow (my time) once the prints are done.

Thanks for the quick response!

fvrmr, 2021-06-15:
We only maintain the appimage of Cura, I will check with a team member to see if they can reproduce this on Linux.

Ghostkeeper, 2021-06-15:
Note that CuraEngine hardly uses floating point calculations, as all of its coordinates and distances are fixed-point. The only places that use floating point are area calculations and some of the angle calculations.

fvrmr, 2021-06-15:
I checked with a team member and they could reproduce. Will bring this up with the team. Keep you posted.

musicCodeGuy, 2021-06-16:
@fvrmr and @Ghostkeeper I can't test it on a raspberry pi 4 because the gpu doesn't support a high enough version of opengl, 

There probably wasn't much point anyway, but it would have been interesting to see if it happened on a 64-bit arm cpu on an arch based distro.

Given Ghostkeeper saying cura mostly uses fixed point math and it works on fvmr's (non linux) machine I'm now suspecting a compiler issue. That'd be a bit strange too given arch linux uses a bleeding edge compiler (gcc 11.1.0-1) and it happens in both the repo packages and appimage, but I have no idea what the appimage binary was built with.

Anyway that's about all I can do.

smartavionics, 2021-06-16:
> @fvrmr and @Ghostkeeper I can't test it on a raspberry pi 4 because the gpu doesn't support a high enough version of opengl,

See https://github.com/smartavionics/Cura/releases for Cura builds for 32 and 64 bit pi4.

nallath, 2021-06-16:
The issue also reproduces for me on Fedora when running it from source. Since I'm running it in debug mode, we can also rule out threading issues.

musicCodeGuy, 2021-06-16:
@smartavionics thanks for that, however your appimage doesn't work on manjaro-arm, it crashes with 

> This application failed to start because it could not find or load the Qt platform plugin "xcb" in ""

An internet search shows it seems to be a well known issue I could probably find a solution for with more searching and fiddling, but I don't have time.

evtrados, 2021-06-16:
For developers, please see internal issue CURA-8321.
I'll also try reproducing it on Ubuntu 18.04 which has glibc v2.27.

smartavionics, 2021-06-16:
I know this doesn't help much but I sliced your project using my Cura build on x86_64 and aarch64 systems and neither showed this issue. Here's the aarch64 result...

[image]

The x86_64 AppImage is built on a Ubuntu 16.04.7 system and the aarch64 AppImage is built on Debian 10.

zlocimir, 2021-06-17:
Exactly same thing is happening with my model when I slice it in Cura, tried yours - same story. I'm also using Ender 5 and Cura is on Manjaro PC built from AUR.
Only for my model layer height doesn't matter since it always show bridge.

[image]

[Kutija-Srednji dio.zip](https://github.com/Ultimaker/Cura/files/6673312/Kutija-Srednji.dio.zip)

smartavionics, 2021-06-17:
Hello @zlocimir, as you didn't provide a project file, I used the settings from @musicCodeGuy 's project. Still looks OK to me...

[image]

musicCodeGuy, 2021-06-18:
So umm with my settings on my machine @zlocimir's STL look the same as his screenshot with the extra bridges. This is with the arch repository cura, as it's preview is WAY faster than the appimage's, so I have no intention of keeping the appimage around unless ultimaker fix the performance issues.

Curiously I've just made a very fancy spiral chess set for my dad, sliced entirely in cura from the arch repos, and printed all the pieces on my ender 5s with no trouble at all. It seems cura doesn't like boxes with holes in them when compiled with particular compiler and/or compiler versions. Yay.

zlocimir, 2021-06-18:
I'm so sorry I totally forgot about a project file, but it doesn't really matter since it also makes this bridge on any of default Ender 5 profiles
[Kutija-Srednji dio.zip](https://github.com/Ultimaker/Cura/files/6679156/Kutija-Srednji.dio.zip)

I don't believe @musicCodeGuy is correct about problem being in Arch user repositories since @nallath reproduced this problem form source on Fedora

musicCodeGuy, 2021-06-18:
@zlocimir The appimage is what in the arch user repository (AUR), it's called cura-appimage-bin, unless you're building from source. that's the appimage binary ultimaker build and support. I and most arch users use the version from the normal arch repository, it's built by arch maintainers and the preview mode is much quicker for some reason. It does exactly the same thing as the appimage ultimaker produce in that it puts these bridges in. Hope that's clear. Fedora (another bleeding edge distro) does it to. Older distros with older compilers don't seem to do this. It's a very strange bug that someone is going to have no fun tracking down.

You model makes the bridges at 0.1mm layer height on my machine using the normal arch packages, haven't tried with the appimage because I uninstalled it.

zlocimir, 2021-06-19:
I'm sorry @musicCodeGuy I got confused about different Arch repositories. I also use Cura from "normal" Arch repository (community). I don't use AUR version that's appimage for the same reasons as you.
I'm really sorry, I got confused.

marcdraco, 2021-07-27:
That's it @Ghostkeeper. Rather amusing since the part I'm designing is part of a project for  (wait for this) Ghostbusters. ;)

Mind you, I do wonder about FreeCAD sometimes - my little trainee did a very rough "robot" yesterday and when we loaded the STL into Blender for checking the mesh was a complete mess even though it looked fine prior to being exported. All on Linux though. (This is Mint 20.2 Uma, based on Ubuntu Focal.) I'll be sure slice the final models on a Windows box. (Now I'll have to go wash my mouth out with soap for using the "W" word.)

richfelker, 2022-04-09:
Any progress on this? I hit it again in a large print where it would have ruined the print, but I live-edited the gcode on octoprint enough to make it at least recoverable. It would be really nice to have this fixed...

richfelker, 2022-04-10:
Being that (I think?) this is only happening for some users with our own builds of CuraEngine, I suspect it's some UB (undefined behavior) that's manifesting in ways that depend on compiler/compiler-version and build options. Reading through the build log, I see a number of warnings from `-Wrange-loop-construct` in code related to polygon outlines. Might this be relevant? I'm not familar with the issue that's reporting but I think it's a case of using an object after its lifetime ends, isn't it?

smartavionics, 2022-04-10:
Possibly related to https://github.com/Ultimaker/Cura/issues/10369 ?

richfelker, 2022-04-10:
Update: fixing those warnings (by using value type rather than reference) does nothing to fix this bug.

richfelker, 2022-04-10:
@smartavionics Thanks for that! Looking at the `simplify` method, I strongly suspect this is an integer overflow issue stemming from multiplying large `coord_t` values or similar.

richfelker, 2022-04-10:
And yes, making `simplify` just `return;` without doing anything fixes my trivial test case (the 100 mm cube from https://github.com/Ultimaker/CuraEngine/issues/1473).

richfelker, 2022-04-10:
OK, here is the buggy (overflowing) condition:

richfelker, 2022-04-10:
OK, here is the buggy (overflowing) condition:

https://github.com/Ultimaker/CuraEngine/blob/7e3a0a5988e291d176a31f594854d8df908f644a/src/utils/polygon.cpp#L532-L537

It's computing a value that's conceptually 128-bit in a 64-bit type: the fourth power of coordinate differences. I'm not even sure the expression is right; if so it might just need some rearrangement to avoid overflow.

smartavionics, 2022-04-10:
Sure, that code could be problematic, but as my version of simplify() uses almost identical code in that spot and doesn't tend to produce the wacky extraneous features that are being discussed here (and elsewhere) it would suggest that there are other issues with the UM version of simplify(). See also https://github.com/Ultimaker/Cura/issues/6748 for further simplify() related badness.

richfelker, 2022-04-10:
@smartavionics I don't want to get into an argument about the other merits of the code. I tracked the bug down to that specific overflow. Here's my stupid patch that does roughly the right computation and makes it work for values that would otherwise overflow:

```
diff --git a/src/utils/polygon.cpp b/src/utils/polygon.cpp
index 9b2fbc83f..9b75907a3 100644
--- a/src/utils/polygon.cpp
+++ b/src/utils/polygon.cpp
@@ -529,7 +529,9 @@ void PolygonRef::_simplify(const coord_t smallest_line_segment_squared, const co
             //h = L / b           [divide by b]
             //h^2 = (L / b)^2     [square it]
             //h^2 = L^2 / b^2     [factor the divisor]
-            const coord_t height_2 = area_removed_so_far * area_removed_so_far / base_length_2;
+            const coord_t height_2 = std::abs(area_removed_so_far) > 0x7fffffff
+                ? area_removed_so_far * ((area_removed_so_far + base_length_2) / base_length_2)
+                : area_removed_so_far * area_removed_so_far / base_length_2;
             if ((height_2 <= 5 * 5 //Almost exactly colinear (barring rounding errors).
                 && LinearAlg2D::getDistFromLine(current, previous, next) <= 5)) // make sure that height_2 is not small because of cancellation of positive and negative areas
             {
```

smartavionics, 2022-04-11:
Very good, I'll consider doing something similar myself. Thanks.

smartavionics, 2022-04-11:
How about this variant? Does it work for you?
```
diff --git a/src/utils/polygon.cpp b/src/utils/polygon.cpp
index b42e1f25..cfb5bd1e 100644
--- a/src/utils/polygon.cpp
+++ b/src/utils/polygon.cpp
@@ -355,7 +355,7 @@ void PolygonRef::simplify(const coord_t smallest_line_segment_squared, const coo
         //Check if the accumulated area doesn't exceed the maximum.
         accumulated_area_removed += current.X * next.Y - current.Y * next.X; //Shoelace formula for area of polygon per line segment.
 
-        const coord_t area_removed_so_far = accumulated_area_removed + next.X * previous.Y - next.Y * previous.X; //Close the polygon.
+        const double area_removed_so_far = accumulated_area_removed + next.X * previous.Y - next.Y * previous.X; //Close the polygon.
         const coord_t base_length_2 = vSize2(next - previous);
         if (base_length_2 == 0) //Two line segments form a line back and forth with no area.
         {
```

richfelker, 2022-04-11:
> How about this variant? Does it work for you?

While I haven't tested it, I believe it's likely to make the symptom go away in the particular cases we tested, but still wrong. It seems possible that `area_removed_so_far` is very large but that the currently-examined edge is very short, in which case even after dividing by `base_length_2` it would still overflow when converted back to `coord_t`. In fact my patch has that problem too.

I think the type of `height_2` also needs to be changed to `double`. Then it should work. Note that @Piezoid's [fork fixing precision issues](https://github.com/Piezoid/CuraEngine/tree/improve-precision) already [does this](https://github.com/Piezoid/CuraEngine/commit/e9460486785ef26882cdb791bc157d2ab804cb41). Note also that (reading those commits) I found there's another copy of the same code with the same bug in `ExtrusionLine` (see https://github.com/Piezoid/CuraEngine/commit/0a32f753d70877561d2dc6abf5f81832bc67f44a) as well as related overflow bugs (see https://github.com/Piezoid/CuraEngine/commit/0e98432805a9a642287e66625a4a86f7d6ca6923).

Piezoid, 2022-04-11:
It is probably best to keep `accumulated_area_removed` and all the shoelace computations as integers. Floating point numbers might accumulate errors when large areas are being added then subtracted. The unit is 1 µm², which is represented with int64s in multiple places in the code, so this should be fine.

Overflows happen when taking the square of `area_removed_so_far` (before dividing by `base_length_2`) The unit is `1 µm^4", requiring twice as many bits. 

> I think the type of `height_2` also needs to be changed to `double`. Then it should work. Note that @Piezoid's [fork fixing precision issues](https://github.com/Piezoid/CuraEngine/tree/improve-precision) already [does this](https://github.com/Piezoid/CuraEngine/commit/e9460486785ef26882cdb791bc157d2ab804cb41). Note also that (reading those commits) I found there's another copy of the same code with the same bug in `ExtrusionLine` (see [Piezoid/CuraEngine@0a32f75](https://github.com/Piezoid/CuraEngine/commit/0a32f753d70877561d2dc6abf5f81832bc67f44a)) as well as related overflow bugs (see [Piezoid/CuraEngine@0e98432](https://github.com/Piezoid/CuraEngine/commit/0e98432805a9a642287e66625a4a86f7d6ca6923)).

That branch is experimental and a bit outdated as I'm waiting for 5.0 to stabilize. Its purpose is to try to reducing the unit size, but that led to more overflows that I had to fix. 

Some of these patches could help hardening the main branch against overflows, even in 1µm precision mode. I started with PRs [#1885](https://github.com/Ultimaker/CuraEngine/pull/1585) and [#1583](https://github.com/Ultimaker/CuraEngine/pull/1583), but won't be merged before Cura's developers clear their 5.0 backlog.

richfelker, 2022-04-17:
Would a simple fix for this be something that could make it in now? This is a real problem that's breaking certain large, simple-geometry prints, which is something I actually come across practical needs for (most recently, a fan duct).

closed 2022-06-08 by casperlamboo

musicCodeGuy, 2022-10-11:
It's been working fine for me since cura 4.9...