Ultimaker/Cura #9980: Material temperature settings are not showing on Cura

SamP3d, 2021-06-15:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

Installed updated version. Restarted cura

### Actual results

No change

### Expected results

Material settings show again

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Have spoken to 3dgbire who are uk dealer of ultimaker machines
[image]

nallath, 2021-06-15:
The Ultimaker 2(+) only set the temperature on the machine itself. As such, you can't set it in the slicer.

closed 2021-06-15 by nallath

SamP3d, 2021-06-15:
It would be very helpful if these settings could be added again to the
Ultimaker 2+ as I am working on a PhD research and needing to alter
temperature settings.
Is this something that could be passed onto Cura Developers?

fieldOfView, 2021-06-15:
You can switch the gcode flavor to "Marlin" (from "Ultimaker 2") via Preferences -> Printers -> Machine Settings. Then you will have full control over the material settings in Cura.