Ultimaker/Cura #9981: Post Processing "Retract Continue" dont work.

Zogar89, 2021-06-15:
### Application Version

4.9

### Platform

Windows 10

### Printer

All

### Reproduction steps

Add Post Processing "Retract Continue".
In my project y put ratio of 1 to be very exaggerated.

When performing travel movements, the extruder does not retract.

[image]

[0h48m20g-Cubo.zip](https://github.com/Ultimaker/Cura/files/6654737/0h48m20g-Cubo.zip)

### Actual results

Extruder does not retract.

### Expected results

Extruder retract on travel.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Zip contains Proyect and Gcode.

fvrmr, 2021-06-15:
Hi @Zogar89 thank you for your bug report.
I will discuss this with the team. Keep you posted!

GregValiant, 2021-06-15:
I don't see it in the Gcode.  This is from the image in your first post.
Line 1931 is the last extrusion in the right rear.
Line 1933 is a **retraction**,
Line 1936 is a **travel move to the left front.**
Line 1943 is a **prime.**
Line 1944 starts extruding in the left front.
Line 1949 is a **retraction**.
Line 1952 is a **travel move back to the right rear.**
Line 1955 is a **prime.**
Line 1956 is the start of extrusion in the right rear.

This is from a slice I did using your 3mf file.
G1 X231.09 Y226.307 E2158.50695        ;Last extrusion in Right Rear
G0 F9000 X233.09 Y226.307                   ;Wipe out 2.0
G1 F2400 E2157.00695                      **;Retraction**
G1 F300 Z11.08                                       ;Z Hop
;MESH:NONMESH
G0 F9000 X27.958 Y40.644 Z11.08          **;Travel to Left Front**
;TIME_ELAPSED:915.924491
;LAYER:39
;MESH:Cubo.3mf
G0 X27.958 Y40.644 Z11.34                      ;Z Move
;TYPE:WALL-OUTER
G1 F300 Z10.44                                         ;Z Hop down
G1 F2400 E2158.71483                       **;Prime**
G1 X67.156 Y40.644 E2162.10453             ;Start to extrude in left front

That looks good too:  Finish extruding in Right Rear, retract, travel to Left Front, Prime, extrude in Left Front.

Zogar89, 2021-06-15:
@GregValiant 
This script should add a retraction movement DURING the travel move.

GregValiant, 2021-06-15:
It does work in "relative extrusion" mode.
G1 X231.09 Y226.307 E3.3897
G1 F9000 X233.09 Y226.307 Z12 E1.3897
G1 F2400 E-1.5
I see no change in the gcode in "absolute extrusion" mode.

Zogar89, 2021-06-15:
> 
> 
> It does work in "relative extrusion" mode. I don't think it's correct though. The E value should be negative? Here it looks like it's extruding more during the move.
> G1 X231.09 Y226.307 E3.3897
> G1 F9000 X233.09 Y226.307 Z12 E1.3897
> G1 F2400 E-1.5
> I see no change in the gcode in "absolute extrusion" mode.

I think that the correct thing in relative mode would be to see that E-1.5 on the same line as the movement.
As found in your example, on another line, perform the retraction after completing the travel move.

Ghostkeeper, 2021-06-16:
> ### Actual results
> 
> Extruder does not retract.
> ### Expected results
> 
> Extruder retract on travel.

To be perfectly clear and make sure we're in line here, the Retract Continue script doesn't add new retractions. It only modifies existing retractions. So line 1948 on your screenshot is not expected to retract.

There are two issues with this script highlighted here. One is a known shortcoming, seen in GregValiant's example. The other one (Zogar's original example) is new afaik.
* RetractContinue does not continue retractions across layer boundaries. This is a known issue, caused by the way that g-code is stored in Cura's front-end and processed by the script. Even if the second issue is fixed, this bug would prevent GregValiant's example from working properly. We've accepted this as a shortcoming so far, and deferred it.
* RetractContinue detects retracted travel moves based on a sequence of `G0` commands. However there is a `G1` command in the middle here for the Z hop, throwing a spanner in the works. So the script sees a retraction, followed by no `G0` whatsoever and it thinks the travel move already ended. This is shown in Zogar's g-code (and reproduced using the project file). As far as I'm concerned, this Z hop should be a `G0` command, not `G1`, since it's meant to be a travel move. Perhaps this shouldn't be considered a bug in the script, but a bug in CuraEngine.

I'll change the title accordingly to match these findings.

Zogar89, 2021-06-16:
The source of this search is that I am printing with a 0.8mm nozzle
During the travel movements I see that there is a small ooze that could be generating these open seam lines.
But I'm not sure as it doesn't happen every time. Still with the same settings. I leave some photos.
Bad seam.
[image]

Good Seam
[image]

Video.
https://photos.app.goo.gl/87f8PDRLLrZ1Mf7E6

Ghostkeeper, 2021-06-23:
> I think this work like: When leaving the extrusion line, it performs a retraction. Then, travel move is as example... 100mm, and ratio is 0.01 = 100mm*0.01= 1mm retract on travel.
When it arrive at yours destination, undo everything. (base retraction + travel retraction).
Is fine?
That is what I have seen happen without jump in Z.

Almost. A summary of the algorithm of this script is as follows:
* Find the next retraction in the g-code.
* Starting from that retraction, collect all of the following `G0` commands.
* Calculate their lengths and reduce the E value by `0.01 * length` at each of those commands.

The unretraction is unmodified, so that it restores the E value to what it was before the retraction (not knowing that this script made that unretraction longer than it was when the unretraction was written).

The problem here lies with the second line. Since the Z hop is a `G1` command, the script thinks that the travel moves have stopped.

In my opinion the Z hop should be a `G0` command. It doesn't need to be performed accurately and carefully. It is also semantically a travel move. So this would have to be changed in CuraEngine.

GregValiant, 2021-06-24:
Everything effects everything.  I have a feeling this one won't be easy.  Retract Continue, Change At Z, Filament Change, Pause At Height.
Fenne better get to work!!

Zogar89, 2021-08-12:
Any news about this?

Ghostkeeper, 2021-08-19:
Actually yeah. This issue was discussed with the team but it was not seen as having great priority. We've had to defer this in favour of more important bugs.

If you or someone wants to have a look in fixing this, I think the issue is with the while loop in line 57 of the script:

https://github.com/Ultimaker/Cura/blob/0348293f87b0a10e9fc2814a5aaa7d994117ca9d/plugins/PostProcessingPlugin/scripts/RetractContinue.py#L57

That loop should not terminate when encountering a `G1` command without `E` parameter.

closed 2023-06-24 by github-actions[bot]

GregValiant, 2023-06-24:
I have added a fix to Retract Continue to disregard the Zhop line so it doesn't short circuit Retract Continue.  #15871