Ultimaker/Cura #9983: Missing support material when rotating

Japhei, 2021-06-15:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Anycubic i3 Mega

### Reproduction steps

1: Use default settings for Anycubic i3 Mega 
2: Load this file: [Support Test.zip](https://github.com/Ultimaker/Cura/files/6657807/Support.Test.zip)
3: If using stl rotate object -15° (blue)
4: Slice

### Actual results

Problem in the given file: Missing support material
[image]
[image]

Problem whith whole file:
[image]
[image]

### Expected results

Same result as without rotation.
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Note: This file runs on an older version (I think 4.8) of Cura without any problem.
File: [Support Test.zip](https://github.com/Ultimaker/Cura/files/6657807/Support.Test.zip)

fvrmr, 2021-06-17:
Hi @Japhei thank you for your report. 
This has to do with the support pattern. If I change it to lines and rotate your part it doesn't show the line correct either. 

[image]

Do you only that line as support? I did have the same result in 4.8 as 4.9.1

If you change your `support overhang angle` to 44 degrees this is the result for better support.  

[image]

This is because your angle is 45 degrees.
[image]

Let me know if that works for you.

Japhei, 2021-06-17:
Hi @fvrmr,
I've printed the same parts in an older Version (I assume 1.8) without any problems, so that's a bit weird.
As I said, the settings are default for the Anycubic i3 Mega.  In this case, that one line of support should be enough. 
Thank you for your workaround, but I think this post should focus on the problem itself. Missing support material looks rather unwanted!
Let me know if you need any more info.

Ghostkeeper, 2021-06-23:
This problem is a bit of a combination of factors.

The lowest cross section of the downwards facing triangular prism happens to be a rectangle with a width of 0.401mm. After insets (1/2 line width from either side) this becomes a rectangle with a width of 1 micron, for the path to follow for the nozzle.

Because of the Maximum Resolution and Maximum Deviation settings, one of the vertices of that rectangle gets discarded, leaving us with a triangle instead. It won't reduce the triangle any further, because that would make the shape completely disappear (the simplification algorithm stops at 3 vertices). This triangle is then offset again with 1/2 line width to get the area that would truly get printed by the printer, with proper miter. This would leave out any parts thinner than 1 line width since those don't get printed. This becomes the area that needs to be supported. The same process is repeated for every of these support layers all the way down.

So basically for each layer you're getting an inset -> simplification -> offset. ClipperLib creates the inset and offset, but does not guarantee that the polygon is rotated so that the first vertex is always at or near the same place as it was in the input. This matters to the simplification in this case, because it stops when it sees that there are only 3 vertices left. So which vertex gets removed depends on where the loop starts. And the result of that influences ClipperLib again for the next layer. This is why the result is inconsistent between the layers.

It all stems from the fact that it's trying to support a polygon which is 0.001mm away from not existing any more. An edge case.

These are some workarounds:
* The setting Use Towers is made to support cases like this better (with a shape that has actual width rather than a 0.4mm line.
* Some profiles use a Support Horizontal Offset of 0.2mm to prevent edge cases like this.
* Using a support line width (and support interface line width) that is lower than the outer wall line width would also guarantee to prevent this case, because the support is still at least as wide as the outer wall line width, otherwise there would be nothing to support. This would leave a support polygon with a width of 50um, which doesn't get removed by the 5um minimum deviation.

Ghostkeeper, 2021-06-24:
I expect that Arachne also solves this issue, by nature of how it would print a 0.401mm rectangle simply with a 0.401mm line rather than a loop.

fvrmr, 2021-06-25:
This issue has been discussed with the team. And we decided to defer this issue, since it is a edge case like @Ghostkeeper explained above. Hope you understand.

closed 2023-06-24 by github-actions[bot]