Ultimaker/Cura #9984: Manage step/mm by filament/setting

Shine753, 2021-06-15:
### Is your feature request related to a problem?

As far as I can see, there is no way to adjust the extruder step/mm by filament.
One can add a gcode command into the printer settings before the start of a print, but this changes depending on the filament you use : step/mm is not the same for one extruder if you use PLA or a soft TPU.
This should be changed into the settings of a print, as well as one can change the extrusion percentage.

### Describe the solution you'd like

Being able to change the value of steps/mm into the print settings. And add a default one into the filament settings.

### Describe alternatives you've considered

Did not find any solution that could be daily used. But this can be done by changing EACH TIME the pre-processing gcode into the machine settings.

### Affected users and/or printers

Everyone who use a printer and search for a good and accurate print quality.

Asterchades, 2021-06-16:
This looks to be pretty much a duplicate of https://github.com/Ultimaker/Cura/issues/9924

nallath, 2021-06-16:
Agreed. This is a duplicate. Closing this in favor of #9924

closed 2021-06-16 by nallath

Shine753, 2021-06-16:
Not exactly, but well, ok. I commented in #9924