Ultimaker/Cura #9986: Gaps in slicing preview Line type

Shine753, 2021-06-16:
### Application Version

4.9.1

### Platform

Windows 7

### Printer

Any

### Reproduction steps

Add a model
Slice it, using line width 0.4 (or 0.48) and Nozzle diameter 0.4.
Display slicing preview in Line type

### Actual results

Between any line, the is a gap. Impossible to know is this is a real gap (it isn't) or a display problem.
This is mainly visible on the top layer

### Expected results

No (non existing) gap between lines

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura.zip](https://github.com/Ultimaker/Cura/files/6659388/Cura.zip)

fvrmr, 2021-06-17:
Hi @Shine753 thank you for your report. 
If you increase your flow to 100% the lines are closer together. 
Let me know if that works for you!

Shine753, 2021-06-18:
That's correct, but this is extremely misleading, as this does not reflect the real extrusion in real world. And does not take into account (and probably can't) any M92 command added in the start gcode commands. I suggest to abandon this behaviour (see discussion about this into issue #9924).

Ghostkeeper, 2021-06-24:
Cura's preview works on the idealised assumption that your printer executes the commands that Cura gives it. If Cura says `G1 E10` then the printer is expected to listen to Cura and move the filament by 1mm.

Similarly, if Cura instructs the printer to print this outrageous overhang, its model expects the printer to execute it faithfully.
[image]

There are print processes that cause the actual print result to be different from this. In my example, the overhang would sag to the point of failure. In your example, it might overextrude. In other examples, you might have warping due to internal stresses, or underextrusion because nozzles are clogged, or blips. Cura doesn't visualise these because it doesn't have enough information on the print process to predict all of it (nor enough processing power and time to model it).

As a catch-all alternative, there is the flow setting. If your settings are tuned well, you won't need it, but a lot of people find it easier to just compensate for slip by increasing the flow rather than figure out why their material is slipping. Cura shows that it would theoretically overextrude then, according to its predictions. 

We consider it out of scope to predict the entire print process for layer view, so I'll reject this feature request.

closed 2021-06-24 by Ghostkeeper